DCC-EX/CommandStation-EX #19: Add callback number syntax to JMRI interface

davidcutting42, 2020-08-28:
Currently, if an ESP32 is connected to the CommandStation and multiple JMRI clients over TCP/IP, there is no way for the ESP32 to differentiate which incoming messages are for which client. I already fixed the problem of messages getting sent out to all connected devices, but that doesn't solve this problem fully because the clients are at a deeper level. Need to implement an optional callback number that is wrapped around the existing JMRI/DCC++ tags. Something like ```[callbacknum<DCC++ tag>]```

ldribin, 2020-08-30:
LarryD added this Issue to the Dev Project

Asbelos, 2020-09-04:
The ONLY module that should know what to do with the reply is the WifiInterface which is common to all wifi traffic. This should NOT be visible to the routines it calls (DCCEXparser, WiThrottle or HTTP callback) That way madness lies.