DCC-EX/CommandStation-EX #33: Release Candidate Feature List

FrightRisk, 2020-09-09:
- [ ] Support Mega, Uno and Nano (some features may not be available on the Uno and Nano)
- [ ] Configuration for Arduino Motor Board and Pololu board
- [ ] Full Throttle Control
- [ ] Turnout Control
- [ ] Sensor Control
- [ ] Output Control
- [ ] Accessory Control
- [ ] All CAB functions F0-F28
- [ ] Network Shield support

### New features ###

- [ ] Throttle reminders
- [ ] Filter / user functions
- [ ] Individual track power control
- [ ] New waveform generator
- [ ] WiThrottle support
- [ ] WiFi Client access to WiThrottle
- [ ] Wifi AP access to WiThrottle
- [ ] New, simpler CAB command
- [ ] Extended <c> command
- [ ] Improved CV read and write

grbba, 2020-09-09:
If you say Network shield support you include the EthernetShield as well ?

FrightRisk, 2020-09-09:
Yes, since that has been supported on DCC++ Classic for a while.

ldribin, 2020-11-04:
This Issue provides the details of what is in CommandStation-EX v3.0.0. and tied to

closed 2022-10-27 by peteGSX