DCC-EX/CommandStation-EX #34: Base configuration decisions

FrightRisk, 2020-09-19:
We need to have a "base configuration" that offers the most wanted features in one setup for a simple install and simple configuration. We can then have one simpler option and as many advanced or specialized options as necessary. What goes in what files for the base configuration and what goes in other files and how does this work with a config.h file?

Optional features and settings:
- [ ] I2C Display
- [ ] Wifi generic settings
      - tx / rx pins
      - use software serial vs. hardware serial
      - port number
      - MAC address
- [ ] WiFi using Ethernet Shield
- [ ] WiFi AP mode
       - Server Name
- [ ] WiFi Client mode
     Static IP or DHCP
      - SSID
      - Password
      - Baud rate
- [ ] Motor Shield Selection
- [ ] Comm Interface type (usb serial, ethernet, wifi)
- [ ] Use trigger pin (debugging)
- [ ] Use EEPROM
- [ ] Enable SHOWCONFIG (turning off saves string space)
- [ ] Show config detail on/off (jumpered pin to ground displays config/debug info on boot)
- [ ] Preamble bit settings (main and prog separate)
- [ ] ACK sample threshold

grbba, 2020-09-19:
In the documentation i would not mention Ethernet Shield in conjunction with WiFi but rather supported WiFi configurations ( e.g. the one from maker labs). EthernetShield refers in general to the cabled version and there is no support for that yet although i am looking into this to add it to CVReader.

Asbelos, 2020-09-19:
Any and every configuration option is an opportunity for someone to get confused and need additional support.. i think some of the above are so obscure, or big-OS conventional  as to be unnecessary in this small machine realtime environment.
MAC address
Preambles (we do enough for an external cutout device... why woukd it be changed?)
Wifi port.(we can only have one port, why not go back to classics one)
Static Ip (do that on your router)
AP name... we base that on the mac. 
CVreader probably doesnt need to save prog space so showconfig could be added to <D CONFIG> Maybe.

Missing from advanced  list  - timer 1-3 in case clashes with some other library.

closed 2020-11-16 by davidcutting42