DCC-EX/CommandStation-EX #36: Feature Request: command to turn the command station into a smart booster (with two districts, or a single higher power district)

FrightRisk, 2020-09-19:
From TrainBoard.

habazut, 2020-09-20:
That would need a different hardware setup because the output must be in sync with the input. I doubt that the Arduino can do that in software. A fast optocoupler is needed at the input as well. To be "smart" it should then include some kind of DCC code sniffer that analyzes the incoming DCC signal and does something useful with it (RailCom?).

Shared source code with the current project "not so much" I'd say.

FrightRisk, 2020-09-20:
Here is Dave Bodnar's version. It isn't DCC++, but can use an Uno/Mega/Pro Mini/Nano... It really just uses an optoisolator to go right into a motor board. Maybe we can do the same. Different firmware to just run a display and cut out the power on overload. Then figure out how to use the motorshield so it looks a lot like a CS, but isn't ;)

Asbelos, 2020-09-21:
Why bother reading the DCC signal through an opto coupler from the track, you can read the DCC signal directly at TTL levels by reading the main track signal pin and feed it directly to the high power motor board(s). 
Handling current detect is going to be the tricky bit with more than one main board... but that can be handled by subclassing the MotorDriver and have it sample multiple pins and maybe have multiple on/off pins.

FrightRisk, 2020-09-21:
It certainly is worth the experiment, but the reason is the distance over twisted pair or shielded network cable you can transmit a TTL signal. In addition to signal loss, there is the potential for a differential in grounds and common mode interference. I can test when I put my O scale between two rooms on the ceiling, but at N scale, I would only have a few feet between the CS and the booster.

davidcutting42, 2020-11-15:
This is something that LCC takes care of. Two of the wires on the bus carry a replica of the DCC signal at exactly 12 volts, and boosters can read off that signal and replicate it. Having a copy of the signal on the LCC bus also allows global RailCom data to be transmitted back to the base station. This is something that I've been including in my new hardware designs. It's pretty cool stuff.

marada1, 2020-11-22:
I have built Dave's booster and have to say I'm pretty happy with how it works. His limits the output to 10 amps but that seems like more then enough. I connected it to the main track out of the motor shield and used the motor shield for the programming track. Seemed to handle DCC instruction to the engine with not problem. Had another gentleman use one of the units I built and powered 4 engines at the same time with no problem. I then purposely shorted the track. Power to the track is shut down until the issue is resolved. Power to main is what turns on the booster.

closed 2025-03-08 by peteGSX