DCC-EX/CommandStation-EX #38: Allow AP mode to be disabled in config file

FrightRisk, 2020-09-20:
A user request is for us to disable SoftAP mode so that there is no AP if they want to use their regular network connection.

habazut, 2020-09-20:
The code in the serial123 branch does not enable SoftAP if it successfully connects to an existing Wifi as "station".

FrightRisk, 2020-09-20:
The comment was made that if it is on, it is transmitting on 2.4GHz on the same channel as a home router (and other routers) so you either deal with the collisions, tie up 2 of the only 3 good channels

habazut, 2020-11-14:
New option in config.example.h: DONT_TOUCH_WIFI_CONF
This will make that the sketch will not try to change anything at startup in the ESP WiFi config and thus the user can configure what he wants with the <+> commands and it will stick to the ESP. So open for always AP mode or static IP or whatever.
Have fun,
Harald.

davidcutting42, 2020-11-16:
Ready to close this issue @habazut or is there more to do?

FrightRisk, 2020-11-17:
Done in 3.0.0

closed 2020-11-17 by FrightRisk