DCC-EX/CommandStation-EX #39: Clarify the network setup and options

FrightRisk, 2020-09-21:
### WiFi Connection Sequence

**Soliciting comments, answers, clarfications, and corrections.**

config.h has the following defines we can use for WiFi:

```
WIFI_SSID "your network name"                   // SSID of the router for Station Mode
WIFI_PASSWORD "your network password" // Password of the router for Station Mode
WIFI_MAC                                                      // Mac address you wish to assign to the ESP8266
WIFI_IP                                                          // If defined, the static IP address, if not DHCP assigns an address
WIFI_PORT                                                    // The port we connect to
WIFI_BAUD                                                   // connection speed for ESP8266. USB baud must be set to match
WIFI_AP_NAME                                            // ????? Is this the "hostname"?
WIFI_USE_SOFTWARE_SERIAL                      // for softwareSerial Setup
WIFI_SERIAL_RX                                           // for softwareSerial setup
WIFI_SERIAL_TX                                           // for softwareSerial setup

```
///// **Question**: aren't the strings, "your network name" and "your network password" a lot of bytes lost in flash?

///// **Question**: what does the "first check" do? What "data may be in the pipeline"?

///// **Question**: If you "shortcut the setup", then how do you change settings? For example, my AP name didn't change change when we changed the software and added the DCCEX_ab48uc8 identifier. I uploaded new firmware, but since it sees I have a setup, it ignores everything and moves past all the setup. We at least need good docs and tech support checklists here because you could get stumped as to why a change isn't being accepted. Must we do AT+RST or something more?

///// **Question**: What about saving settings in EEPROM as was posted on Discord?

///// **Question**: How does softwareSerial impact this?

1. Activate the first serial port (Serial1) and set its speed to WIFI_BAUD. Check to see if we get a response from the WiFi hardware

2. If we get a response, ???????? what happens?

3. If there is no response, try Serial2 and if still no response, try serial3

4. If no response after serial3, ?????? What happens?

5. If we get a response, set the WiFi hardware to station mode and check to see if it has been configured by looking at the WIFI_PASSWORD string to see if the defalt, "your network " is still in the password setting. 

6. If there is no configuration, then check to see if it may have already been configured and is using settings saved in the WiFi device to try and connect to a previous network. We do this by waiting 8 seconds and then sending the command to check for an IP address.

7. We then check to see if there is a MAC? And if we have one? What if we don't?

8. If there IS a valid IP address already we skip to 10

9. If there is NO valid private IP address (indicating we connected to an AP as a client/station), we attempt to setup the WiFi device as an AP.we check to see if the WiFi device has older or newer firmware. 

  a. If we detect OLD firmware, we check to see if there is an SSID set, and if so, we use the SSID
     and password to try and connect to the local network router. We then ALSO try to setup a software 
     AP mode creating an SSID of DCCEX_123456 and PASS_123456 where "123456" is the last 6 characters 
     of the MAC.
  b. If we detect NEW firmware, we set the hostname, check if there is an SSID set, and if so, use the
     SSID and PW to change the CURRENT settings and connect to a local network router. We then ALSO setup
     a soft AP with an SSID of DCCEX_123456 and PASS_123456 where "123456" is the last 6 characters 
     of the MAC. Finally, we set the TCP recieve mode to "active mode", which sends all data instantly to
     the Command Station without being held in a buffer.

///// **Question**: What is the difference between SSID and Hostname as it relates to what we see on our network
                list and ED? Is only "hostname" available to substitute for IP address in something like ED?

10. We set a TCP server to be able to handle multiple connections
11. We enable the TCP server
12. We display the IP addresses and MAC addresses for the Station (Client) and AP

///// **Question**: We are calling this in the main .ino sketch. Should more of this be done inside the library? In particular the checking off each serial port to find a device? I see why it works here, devices could have different numbers of serial ports. But thoughts?

 
```
Serial1.begin(WIFI_BAUD);
  wifiUp = WifiInterface::setup(Serial1, wifiESSID, wifiPassword,dccex,port);
#if defined(ARDUINO_AVR_MEGA) || defined(ARDUINO_AVR_MEGA2560)
  if (!wifiUp) {
    Serial2.begin(WIFI_BAUD);
    wifiUp = WifiInterface::setup(Serial2, wifiESSID, wifiPassword,dccex,port);
  }
  if (!wifiUp) {
    Serial3.begin(WIFI_BAUD);
    wifiUp = WifiInterface::setup(Serial3, wifiESSID, wifiPassword,dccex,port);
  }
#endif
```

habazut, 2020-09-21:
> ///// Question: aren't the strings, "your network name" and "your network password" a lot of bytes lost in flash?

If you feel that, figure out something that is shorter and can be understood as well. The compared 13 chars are in RAM
because how strncmp_P() works (yes, I added a library function to do that).

> ///// Question: what does the "first check" do? What "data may be in the pipeline"?

Just because you reset the Arduino does not mean the ESP is reset. It
can be in the middle of a receive.

> ///// Question: If you "shortcut the setup", then how do you change settings? For example, my AP name didn't change change when we changed the software and added the DCCEX_ab48uc8 identifier. I [...]

If you change the strings "Your ...." then it should force out that to the ESP.

> Must we do AT+RST or something more?

AT+RST does not change what you did set as SSID and Password.

> ///// Question: What about saving settings in EEPROM as was posted on Discord?

Yes, that might be an idea but noone has written the code yet and I think we can release as is.

> ///// Question: How does softwareSerial impact this?

As good or as bad ;-)

>     Activate the first serial port (Serial1) and set its speed to WIFI_BAUD. Check to see if we get a response from the WiFi hardware
> 
>     If we get a response, ???????? what happens?
> 
>     If there is no response, try Serial2 and if still no response, try serial3
> 
>     If no response after serial3, ?????? What happens?

Not quite like that. It just tries to start Wifi on each Serial{1,2,3}.

The following happens in each try...

>     If we get a response, set the WiFi hardware to station mode and check to see if it has been configured by looking at the WIFI_PASSWORD string to see if the defalt, "your network " is still in the password setting.
> 
>     If there is no configuration, then check to see if it may have already been configured and is using settings saved in the WiFi device to try and connect to a previous network. We do this by waiting 8 seconds and then sending the command to check for an IP address.

>     We then check to see if there is a MAC? And if we have one? What if we don't?

Hm. I should catch that but it would be strange if the AP did not have a MAC as it must have one by design. We could check that all chars of macTail are within a-f0-9

>     If there IS a valid IP address already we skip to 10
> 
>     If there is NO valid private IP address (indicating we connected to an AP as a client/station), we attempt to setup the WiFi device as an AP.we check to see if the WiFi device has older or newer firmware.
> 
> a. If we detect OLD firmware, we check to see if there is an SSID set, and if so, we use the SSID
> and password to try and connect to the local network router. We then ALSO try to setup a software
> AP mode creating an SSID of DCCEX_123456 and PASS_123456 where "123456" is the last 6 characters
> of the MAC.
> 
> b. If we detect NEW firmware, we set the hostname, check if there is an SSID set, and if so, use the
> SSID and PW to change the CURRENT settings and connect to a local network router. We then ALSO setup
> a soft AP with an SSID of DCCEX_123456 and PASS_123456 where "123456" is the last 6 characters
> of the MAC. Finally, we set the TCP recieve mode to "active mode", which sends all data instantly to
> the Command Station without being held in a buffer.
> 
> ///// Question: What is the difference between SSID and Hostname as it relates to what we see on our network
> list and ED? Is only "hostname" available to substitute for IP address in something like ED?

I am not sure HOSTNAME shows up anywhere, the AT+CWJAP_CUR matters IMHO.

> 
>     We set a TCP server to be able to handle multiple connections
>     We enable the TCP server
>     We display the IP addresses and MAC addresses for the Station (Client) and AP
> 
> ///// Question: We are calling this in the main .ino sketch. Should more of this be done inside the library? In particular the checking off each serial port to find a device? I see why it works here, devices could have different numbers of serial ports. But thoughts?

Yes, but I did not have a better idea/code yet.

davidcutting42, 2020-11-16:
@habazut @FrightRisk where are we at with this issue?

davidcutting42, 2020-11-17:
From #55, which I just closed as duplicate:

> The current WiFi documentation does not tell the user how to get the password for the AP, nor how to customize the password. Overall, the documentation is very wordy and needs to be cleaned up. It should probably be split into two files: a basic setup and an advanced setup file.