DCC-EX/CommandStation-EX #41: Figure out how to set motor board in config file

FrightRisk, 2020-09-24:
The code that uses a define to create an alias for ARDUINO_MOTOR_BOARD works just fine and compiles on everything except the Github check. Let's find a better way to do this then.

dexslab, 2020-09-24:
@FrightRisk This was already fixed im not sure what issue you are having but it did work just fine

FrightRisk, 2020-09-24:
Code compiles in Arduino and PIO. Code works just fine. Not sure why GitHub has a problem with it. Do you? I was trying to set the motor board in config.h instead of having someone go to two files to config things and change a macro in a function. Maybe there is a better way. Frankly, assigning numbers to motorboards, network types, etc seemed fine in Classic. You just enter a 0 or a 1 or a 2. Less room for a non coder to make a mistake.

ggee, 2020-09-24:
Isn't the motor board already in the config.example.h in the config branch?

#define MOTOR_SHIELD_TYPE STANDARD_MOTOR_SHIELD

I did notice the ino file is referencing a variable that doesn't exist.

  DCC::begin(MOTOR_BOARD);

So didn't compile in PIO for me.

dexslab, 2020-09-24:
To set motorshield type you use the symbols that are already defined on line 27 there is a full list above that line

FrightRisk, 2020-09-24:
Sorry guys, I forgot you don't have my config.sys!!  No wonder it isn't working. I exclude that in the .gitignore. so I imagine the display won't work either. I guess I can rename it and upload it and you can rename it. If in the config.sys, you have define MOTOR_BOARD MOTOR_SHIELD_TYPE then it will work. Because you can set type to STANDARD_MOTOR_BOARD

davidcutting42, 2020-11-16:
@FrightRisk is this issue ready to be closed?

FrightRisk, 2020-11-17:
Implemented now in config.example.h which is the template for a config.h file.

closed 2020-11-17 by FrightRisk