DCC-EX/CommandStation-EX #48: Use IO pins for turnouts

matt-hu, 2020-11-01:
It would be useful if it was possibly to set up a turnout that was controlled by the IO pins of the Arduino via something like a relay.

FrightRisk, 2020-11-02:
We already do that, unless I misunderstand you. https://github.com/DCC-EX/CommandStation-EX/wiki/DCC-EX-Command-Reference#arduino-output-pins

matt-hu, 2020-11-02:
@FrightRisk I understood that you could use IO pins for sensors, but I hadn't seen the section on outputs. 

This got me thinking, would there be a way to map IO pins to 'virtual' accessory decoders so that software that sends the `<T>` command can still activate turnouts connected to the IO pins?

davidcutting42, 2020-11-16:
@matt-hu I believe you can probably do this by writing a custom command filter. @Asbelos might have some more info.

Asbelos, 2020-11-16:
The issue here is the drag of backward compatibility. DCC++ was originally designed to be run from JMRI and the implementation of turnouts was awkwardly split between DCC turnouts which the command station knows about  and other outputs or accessory packets sent by JMRI  which the command station cant tell the difference between a turnout or a cow on the line!So if you are using JMRI there is a way already... otherwise you will have to wait for the next version where we need a major overhaul of this mess.

closed 2020-11-16 by davidcutting42

matt-hu, 2020-11-17:
@Asbelos thank for the help.