DCC-EX/CommandStation-EX #49: Remove SAMC21/SAMD21/ATMEGA4809 support

davidcutting42, 2020-11-15:
Because FireBox is no longer with us, and I'm the only one that has any SAMC21 based hardware, it is best to remove the SAMC21 from the codebase to keep things simple.

Along the same lines, nobody is using SAMD21 (except maybe @mstevetodd - please confirm) so it will be easier to just remove that from the codebase.

Same for the ATMEGA4809 (@PilotinControl25 can you please give a status update here?)

The extra processors can be put into a new branch and held for later use

PilotinControl25, 2020-11-15:
Uno wifi rev2 is still in play and the new metro wifi from Adafruit is
coming out....these should not be overlooked

davidcutting42, 2020-11-15:
This is true, but the boards are not currently officially supported, and the priority is on documentation right now. I recommend removing them for this release to another branch and adding support back in for future releases. @FrightRisk can you please comment?

FrightRisk, 2020-11-15:
I didn't understand them to be for this release. 3.0.0. Has Uno, Nano, Mega (focus on Mega) 3 or 4 Motor Boards and some WiFi options and a Network Shield.  Uno and Mega WiFi need testing. So they are on the list for evaluating after this release.

mstevetodd, 2020-11-15:
I'm still playing with my FireBox, but I understand why you'd remove the code. I'll figure out a way to keep it in my local copy.

PR #50 (2020-11-16): Remove all boards but Uno, Mega, and Nano
Closes #49 - also created another branch to maintain old files
closed 2020-11-16 by davidcutting42