DCC-EX/CommandStation-EX #56: Clean up files

davidcutting42, 2020-11-17:
Currently there are *too many files* in the root of the repo. @Asbelos what is your opinion on moving them to subfolders? In the Arduino IDE recursive compilation is only done on the ```src``` folder. Can put the non-config and non-ino files in there? If you want to take it a step further we could divide into even finer folders within ```src```.

Asbelos, 2020-11-17:
Moving code to src subdirectory hides it from the Arduino IDE ...means you cant edit or view it in the IDE.   MAYBE that's a good thing but on balance i think not.

FrightRisk, 2020-11-17:
Not now. Let's focus on a stable system for a bit.  We would also have to put the explicit paths into each file when we are trying to get a release out this week.

closed 2020-11-17 by FrightRisk