DCC-EX/CommandStation-EX #61: Develop WiThrottle Integration plan

FrightRisk, 2020-11-18:
1) Use of Withrottle to move stuff about while automation is paused, get stuff into starting positions, run track cleaner train,  etc etc
2) TPL to hand over a loco for shunting, and ED to hand it back by sending it on a TPL route.

There are also implications for things like turnouts because TPL cannot operate on turnouts dynamically created by JMRI or ED. In effect, the layout wiring needs to be known in advance for an automation to work. Thus the layout definition (Turnouts, sensors, outputs)  is effectively coded in a file that is included at compile time. [Remember the purpose of TPL is to automate within the CS without JMRI.]

May be able to: 
- intercept the Turnout aspects of Withrottle to call into the TPL layout handling. 
- Intercept loco acquisition so that it cant steal a loco TPL is animating.
- Use Routes to allow ED to send a loco onto a TPL route.
- Pre populate a throttle when handling a loco over from TPL..... BUT not sure how to handle multiple ED clients here.

mstevetodd, 2020-11-19:
My initial thoughts on your list:
> intercept the Turnout aspects of Withrottle to call into the TPL layout handling.
As long as the server sends a list of turnouts and state updates, it can define and use them however it wants.
> Intercept loco acquisition so that it cant steal a loco TPL is animating.
Client waits for Mx+ response to its Mx+ request. So server can simply choose not to respond, or (better) send only a message explaining the issue.
-OR- (better?) Server can reply with Mx+, but then ignore any speed/direction requests while server is in control.
> Use Routes to allow ED to send a loco onto a TPL route.
Same as turnouts, client does not care what is behind the list.
> Pre populate a throttle when handling a loco over from TPL
Client should accept unsolicited Mx+ messages and add that addr to throttle "x". ED and WiThrottle should always have throttle "0" visible (?). 
> BUT not sure how to handle multiple ED clients here.

This seems simpler if the flow looks something like this:
1) Client requests loco address AAAA for a specific throttle "x"
2) Server responds with Mx+AAAA
3) Client selects a "Route" which indicates server should control AAAA
4) Client still shows AAAA, but any client speed/direction requests are ignored with message "Loco AAAA running under TPL"
5) Server completes task and stops ignoring speed/direction for AAAA.
This removes the need for the server to "guess" at which client and throttle.

Asbelos, 2020-11-19:
Good ideas. Makes sense, thanks.

Client requests loco address AAAA for a specific throttle "x"Server responds with Mx+AAAAClient selects a "Route" which indicates server should control AAAAClient still shows AAAA, but any client speed/direction requests are ignored with message "Loco AAAA running under TPL"Server completes task and stops ignoring speed/direction for AAAA.

closed 2025-03-08 by peteGSX