DCC-EX/CommandStation-EX #65: Layout technology general

Asbelos, 2020-11-18:
TPL requires access to the layout control hardware (turnouts sensors, outputs, signals) and is not compatible with the JMRI way of defining this externally and downloading it at run time... Quite apart from the fact that this severely limits the layout complexity and quickly overflows the EEPROM.

Current TPL code expects a 
  LAYOUT
    ENDLAYOUT 
 block containing definitions of hardware devices. This information is stored in progmem and only the current status is mapped into bits in a fixed byte array which can be readily stored in eeprom.

It seems to me that it would be entirely reasonable to allow this layout definition in a non-TPL build - as an alternative to the JMRI methods which are fundamentally incompatible with TPL.

Asbelos, 2020-11-18:
If this were built in by a user, then the WiThrottle and <T id 1|0> commands would continue to operate but the define/delete versions would be denied. 

Sensors also behave slightly differently as well... If JMRI is using sensors then they still have to be continuously polled, whereas TPL does not need this overhead.

So... We have I suggest, 3 different types of CS build
 1. As now with all the old stuff
 2. With added compile time Layout definition. {define/delete versions of commands denied)
    [The presence of a myLayout.h file would be enough to switch this mode on] 
 3. With added compile time layout and TPL Automation.   
    [The presence of a myTPL.h file would be enough to switch this mode on]

Asbelos, 2020-11-18:
If We do implement this... can we avoid the hassle of rewriting the old eeprom code and just leave it as-is?

matt-hu, 2020-11-18:
@Asbelos I haven't really done any work on the Command Station side of things before, but the way I see it, JMRI poses a major problem for adding new features as it requires DCC++ to work a specific way. I could be wrong, but I am wondering if it would be better off in the long run if someone could add a new 'driver' in JMRI, specific to Command Station EX, allowing more freedom development wise.

On the TPL side of things, am I right in saying that the scripts are compiled in with the command station then uploaded to the Arduino? Even if this is so, it could be all right to define the turnouts in the script, as most people would only add turnouts to their layout very occasionally. Also, as TPL is very much abstracted from C++, do you think it would be possible to have an interpreter that runs on a PC and sends commands over serial for testing purposes? Once the script had been finalized, it could then be compiled.

As I said, I don't know a huge amount about Command Station or TPL so these suggestions may be no good, but I thought I would say them just incase they were helpful in any way.

Asbelos, 2020-11-18:
I tend to agree with your points about JMRI although I am not familiar with the code this would involve.The idea of debugging Tpl on a pc is viable but would involve a fair bit of code.

matt-hu, 2020-11-18:
@Asbelos I am happy to have a go at writing the PC application once a full command list is made. I was intending to do this anyway at some point as I wanted to support TPL execution over Wifi on [TrainLink](https://github.com/trainlink-org/trainlink-api).
I also don't know much about JMRI, but if I have some free time I will have a little look to see if anything can be done.

Asbelos, 2020-11-18:
I suggest taking a look at this doc that has the list... but beware things can change a bit as we progress the development. https://github.com/DCC-EX/CommandStation-EX/blob/TPL/docs/TPL%20Introduction.docx

matt-hu, 2020-11-18:
@Asbelos thanks for that!

closed 2025-03-08 by peteGSX