DCC-EX/CommandStation-EX #66: Command Station 3.1 Release Candidate Features

FrightRisk, 2020-11-19:
### New Features for 3.1 ###

- [ ] TPL Automation
- [ ] New WiThrottle Integrations
- [ ] New JMRI integrations
- [ ] Rework accessory <a> command for 1 part linear addressing option
- [ ] Create better EEPROM storage format
- [ ] Add SDCard support option
- [ ] Extended <c> command to provide current in reading, mA and show trip value and max value
- [ ] Consisting
- [ ] V Command (verify CV aka validate)
- [ ] WiFi credentials override and not saved in sketch for security
- [ ] Allow DCC accessory messages over PROG track
- [ ] Web Throttle onboard the CS
- [ ] Diags through WiFi/Ethernet
- [ ] Async prog track commands over WiFi/Ethernet
- [ ] DIAGs through WiFi/Ethernet
- [ ] Turnout processing/Better class (may need a filter to handle WiThrottle integration)
- [ ] Turnout names? ex: "coal yard"
- [ ] Extended turnout command to handle servo/stepper angle?
- [ ] Loop speed detector
- [ ] Off board accessory drivers API
- [ ] Command to automatically handle long and short writes
- [ ] Customizable LCD display
- [ ] Use ring buffer to broadcast messages from WiThrottle such as power status, speed, direction, turnout state
- [ ] Implement frame rate monitor
- [ ] Finalize use of <U> Command
- [ ] Add a command to JMRI and the CS to report a list of device capabilities (devCaps structure)
- [ ] Add loconet support? (use their throttle)
- [ ] Add USB "tether" feature to WiThrottle so we can use a USB cable connection without WiFi
- [ ] Turn PROG track into a "booster" so when not in use, it can be a separate power district

### Bus support ###

- [ ] evaluate LCC, RF24, MQTT, etc. 

### Installer ###

- [ ] Add full cross-platform support
- [ ] GUI Enhancements
- [ ] Built-In serial monitor
- [ ] WiFi support

### WebThrottle-EX

- [ ] Save Turnouts (needs new data structure)
- [ ] GUI enhancements

### Hardware ###

- [ ] Railcom support
- [ ] FireBox/FireBit
- [ ] Mobile throttle

Asbelos, 2020-11-20:
Loop Speed and Frame Rate items in list are the same thing I think.
Missing is a performance hike by shortcutting the channel B cycle if power is off. (DONT do this for the A cycle or bad things will happen.)

Does each of the ifeatures need to be on separate cards so that they can be individually discussed/tracked?

davidcutting42, 2020-11-21:
@Asbelos I think they should be on separate cards. I'll take care of that now.

davidcutting42, 2020-11-21:
Checking off items as they are moved to separate cards.

closed 2020-11-21 by davidcutting42