DCC-EX/CommandStation-EX #74: Add SD card support option

davidcutting42, 2020-11-21:
Might host turnout/input/output data, and/or host an instance of exWebThrottle.

matt-hu, 2020-11-21:
Having an instance of exWebThrottle onboard requires a webserver, something not available without programming the ESP8266, as far as I am aware. Also, the ESP8266 might not be the _fastest_ web server ever... may need an ESP32.

Asbelos, 2020-11-21:
Web server speed will be pretty irrelevant when you only have to serve a handful of static files once to each of a handful of clients.

matt-hu, 2020-11-21:
@Asbelos very true, although I think the major flaw is having to reprogram the ESP every time we release a new version of exWebThrottle. (for more info on exWebThrottle distribution, see DCC-EX/exWebThrottle#15)

davidcutting42, 2020-11-21:
> the major flaw is having to reprogram the ESP every time we release a new version of exWebThrottle

@matt-hu not if you have an SD card. Then you can just have it serve static files from the SD card. 

I also have a hard time believing that we can't serve exWebThrottle over AT commands. Might be a little hacky, but if you look around with a quick google search there are ways.

matt-hu, 2020-11-21:
@davidcutting42 Ok, those are good ideas...
I will look into the AT command stuff, but I also think that PWAs should be used as well, so users have a choice.

ManiAkasapu, 2020-11-22:
Come up best idea Gents..:)

davidcutting42, 2020-11-22:
@matt-hu I agree that the ideal option would be to port everything over to the ESP32 and serve the webpage up from there. Unfortunately, the ESP would require a lot of changes and bloat in the current firmware. We are considering the option of moving to the ESP32 as our only platform for the v4 release and maintaining backwards compatibility with v3.

closed 2025-03-08 by peteGSX