DCC-EX/CommandStation-EX #75: V Command (verify/validate CV)

davidcutting42, 2020-11-21:


Asbelos, 2020-11-24:
This code is already in master, but not presumably in Documentation and I don't know if kit has been tested by anyone.
IMHO this should probably be removed for the reasons below:
 
Command is <V cv value>  or <V cv bit 0|1>

The intended purpose was to improve CV writing performance where it is expected that the cv is likely to already contain the value. However I am dubious as to the advantage because writing a cv  is  WRITE/ACK/VERIFY/ACK where both acks are expected to be true and thus quite fast whereas the V command would be VERIFY/ACK then maybe [WRITE/ACK./VERIFY/ACK] with the possibility that the first ACK may be false and thus relatively (like 10 times) slow.... so... a small gain if the value is already there vs a large loss if it isn't!

Asbelos, 2020-11-24:
Damn markdown...   commnnd is ``<V cv value>``  or ``<V cv bit 0|1>``

closed 2021-05-19 by Asbelos