DCC-EX/CommandStation-EX #79: Create better EEPROM storage format

davidcutting42, 2020-11-21:


matt-hu, 2020-11-21:
Would it be possible for a list of cabs to be stored in EEPROM if turnouts and outputs are going to be compiled with TPL? This would mean that they only have to be set up once, so if a user was using, say, JMRI and exWebThrottle, as long as both supported it, then they would both know about the cabs because the CS would tell them?

Asbelos, 2020-11-21:
Tpl will need eeprom to store cab state for restarts etc.. its incompatible with jmri. If you do run jmri then you must connect withrottles through that and not directly.

grbba, 2020-11-22:
Actually I think we also have to look not only to the storage format but also space allocation if some wants to use also EEPROM for his/her own purposes either being through own development or by using other libraries I guess so we have to tell others you can use this space from here to here or something like this ... ?

Asbelos, 2020-11-23:
Eeprom is very small, we could reserve x bytes at start for user data but x will have to be very small.

closed 2025-03-08 by peteGSX