DCC-EX/CommandStation-EX #84: Off board accessory drivers API

davidcutting42, 2020-11-21:


grbba, 2020-11-22:
Don't know exactly what it means but brings up a question for me. We should build first a complete architectural view about a cheap easy to use DCC ecosystem. We can't / shall not support everything from the start ( and even later ) but make some opinionated choices towards the easy/cheap goal. We should actually be able to tell users what bricks to get how to stick them together and if you do this here is what you can do. We support this and make sure hat it works. so if we decide e.g. all accessories will run through lets say RF24 modules using protocol xyz thats it nothing more nothing less.  There will be the perfect guide for doing this ( its just an example I didn't mean that as a choice ). If we try to cover all and everything nothing will be good to the level users would expect it to be looking at the ressources available.

marada1, 2020-11-22:
This makes sense. Get what your working on now going and bulletproof. When there is a product that the masses like and want then look to that second product like the ESP32 system. That will not appeal to as wide an audience.

Neil-McK, 2021-05-05:
Hardware abstraction layer under development may help with this.  In this model, remote devices attached to GPIO extenders and DCC accessories are treated the same as local Arduino pins, both for sensors (input) and for turnouts and outputs.  So setting a remote digital output for an LED, or operating a servo to move slowly between two positions, is requested by a <Z n 1> or <T n 1> command.
This can be extended to LCN devices and others.

Neil-McK, 2023-02-16:
The Hardware Abstraction Layer (HAL) has now been a core part of the CS for over a year and has platform-independent drivers for various GPIO Extenders, servo and LED PWM drivers, OLED and LCD displays, turntables, Arduino-based GPIO Expander, Rotary encoders, analogue inputs and more.  Also, network support has been prototyped to enable device states to be send/received over ethernet, nRF24, and other media.

closed 2025-03-08 by peteGSX