DCC-EX/CommandStation-EX #85: Command to automatically handle long and short writes

davidcutting42, 2020-11-21:


Asbelos, 2020-11-22:
Eg command <W 1234> would write locoid 1234 to the loco on prog track and automatically handle the correct cvs and status bits for the long or short form. (The write equivalent of <R>)

grbba, 2020-11-22:
That is ok for me ( and i implemented that in my RestAPI using basic write commands available ) but poses in general the question about what i would call 'higher order functions' and how to expose them to the user. As an comparison i would say the current JMRI (DCCEX) commands are like Assembler and a function like this is already on an abstraction level like 'C'. So for this type of function should we stick to the Assembler format for exposing them ?

Asbelos, 2020-11-22:
Reading a cv is already a high order function involving 10 separate conversations with the decoder... reading a cab address is already an 11 to 21 conversation function. It just seems that writing a cab address, at 2..3  conversations  is missing...

Asbelos, 2021-01-17:
Remember this should also zero any consist address to avoud confusion.

Asbelos, 2021-01-18:
ConsistR branch contains this command < W locoid >

Asbelos, 2021-01-21:
Merged to master 3.0.3

closed 2021-01-21 by Asbelos