DCC-EX/CommandStation-EX #86: Customizable LCD display

davidcutting42, 2020-11-21:


5chufti, 2020-12-30:
at least solve contradiction between configurable lines/columns in config.h (user) and fixed settings in LCDDisplay.h, most 4 line displays have 20 columns...
best would be to take over definitions from config.h

Neil-McK, 2021-03-02:
Whilst the 4-line LCD has 20 characters, and 128-column OLED displays are capable of 21 characters, 2-line LCDs are only 16 characters long.  So to avoid losing the end of messages off some screens, we limit the length of the messages to 16 characters. 
A second option would be to drop support for the LCD1602 display, allowing messages of up to 20 characters.
A third option might be to code the messages so that they can be abbreviated when on a 16-character screen, e.g. by marking optional characters with a prefix character. This would make the messages slightly longer in memory.

5chufti, 2021-03-02:
as I allready wrote: best would be to take over the setting from config.h as there the user has to/is expected to put the actual values of his display after all...
then it is up to the implementation of the "LCD" macro to prevent buffer overflows (cut chars/lines) if the passed text would exeed the defined chars/lines

Neil-McK, 2021-05-05:
Parameter MAX_MSG_SIZE can be used to allocate buffers for longer lines.  This defaults to 16 since all existing messages are designed to fit on a 16 character LCD, so longer buffers would be wasted RAM, particularly on an Uno.

Example for a 132x32 SH1106-based OLED display:
```
#define MAX_MSG_SIZE 21
```

closed 2021-05-05 by Neil-McK