DCC-EX/CommandStation-EX #95: Custom AT firmware

davidcutting42, 2020-11-22:
I would be interested in developing custom ESP8266 AT firmware, based on the current source code, that could provide the following:

- A server for WebThrottle-EX
- Real mDNS support
- Possibly provisioning support in AP mode.

Any other capabilities that we would want? This is more of a stopgap solution before (hopefully) the ESP32 replaces the entire arduino/esp8266 mess come v4.0.0

Obviously, we would have to have an automated program that can set up the ESP with new firmware using the Arduino as a passthrough, and two (maybe just one) extra jumpers would be required.

davidcutting42, 2020-11-22:
And just to be clear, the CS would still be compatible with stock AT firmware. Additional commands would be extras that could be discovered by the CS using a "discover" AT command

Asbelos, 2020-11-22:
Its likely that custom firmware would be incompatible with existing wifi libraries as we would have multiple processes sending or listening on the wifi serial. 

We could manage this if we control all AT traffic as we do now but this woukd conflict with @grbba s networking rewrite.

FrightRisk, 2020-11-22:
That one firmware can do AT command OR networking commands, correct? When
you say based on existing firmware, are you talking about the one that
Atani wrote in BaseStation-EX?

grbba, 2020-11-22:
No you can't have both at the same time. Actually the library i use is based on the AT 1.7 or AT 2.0 firmware and then in the library provides the **same API** as the Ethernet library which make its use much simpler. The WiFi library then adds a lot of additional features such as AP setup delete prexisting settings etc etc ...

grbba, 2020-11-22:
@davidcutting42 Just for clarification ... What do you understand by firmware ? Difference with a sketch ? If the whole CommandStation would run on an ESP the only difference for me would be that i wouldn't have to rely on the current WifiEspAT library. The ESP Wifi library has the same API as the Ethernet one so no real code changes i would say except refactoring some class names i'd say.

Asbelos, 2020-11-22:
If we stay with the CS wifi code we have now, then it's possible to integrate new AT commands into the ES code and handle them from the CS. if we plan on using a wifi library like adafruit on the CS, then we cant add AT commands without also modifying the library.

davidcutting42, 2020-11-22:
Yeah, so just to be clear, this is an extension of the AT firmware. We would basically be adding extra AT commands for things like taking care of the web server, etc. So it would be the same API as the current AT command set, with some extra commands. Am I making any sense?

Asbelos, 2020-11-22:
If you are then going to use the standard wifi library on the arduino, you won't have a path down which to send your new AT commands.

FrightRisk, 2020-12-07:
So we wrote our own ethernet too possibly. Has anyone looked at the WIFI code in the esp8266 folder of the previous version of DCC++ EX? That uploads to a thingie

Asbelos, 2020-12-07:
It appears to me that the code is large and complex because it parses and recomposes <> commands sent from wifi clients before passing them on to the arduino. This suggests to me that it would be complex to maintain as every new command would have to be added to this code as well as the arduino code. Presumably this also doesnt support the withrottle protocol either.Apologies to Atani if I've got the wrong end of the stick. I appreciate the wish to make the wifi handling easier and build in web pages etc but we still have to consider the real world complexities of building,  documenting, maintaining, installing and supporting vs the anticipated gain to the end users.

closed 2025-03-08 by peteGSX