DCC-EX/CommandStation-EX #103: Bug in WifiInterface.cpp

Neil-McK, 2020-12-10:
I'm really impressed with the work that has been done on this project.  It's a big improvement on the already functional and useful DCC++ software.  However, I did encounter a few problems when I tried to get the WiThrottle component to work.  The initial problem was that my throttle (which is based on an ESP8266 with OLED screen, battery and rotary encoder) wouldn't work with the DCC-EX WiThrottle server, whilst it works fine with JMRI.  This was eventually tracked down to fragmentation of the frames in the transmitted TCP stream;  the partial TCP implementation in the DCC-EX server treats TCP like UDP and does not reconstitute the TCP stream from the IP packets.  I was able to fix this by changes in my throttle to send commands in a single command (and hence a single IP frame), so this isn't particularly critical.

While inspecting the code to understand what was going on, I did spot a small bug in WifiInterface.cpp:

- Line 236 calls checkForOK() to read the response string "+CIFSR:APMAC" ;
- The following lines then gobble the mac address bytes that follow; 
- This leaves the trailing "OK" in the input  buffer, unread;
- When the next command "AT+CWSAP" is sent, this residual "OK" is immediately treated as the response.  
- Consequently, when the "AT+CIPRECVMODE=0" is sent, it is rejected with an error as the AT+CWSAP has not completed yet.

This will prevent the WiThrottle server from working if the receive mode is incorrect.

The bug can be resolved by adding a call to checkForOk() after the string macTail has been assigned (or before, if preferred).

Keep up the excellent work!

FrightRisk, 2020-12-10:
Excellent! Thank you for the report.

FrightRisk, 2020-12-27:
@Neil-McK Want to take a crack at how you would reconstitute the stream from the IP packets? ;)

habazut, 2020-12-27:
the checkForOk() should be an easy one. The tcp frag issue is probably a tougher one as it may depend on ESP firmware.
Harald.

Neil-McK, 2020-12-28:
The reason for the fragmentation was my rather naive way of sending the commands over TCP using multiple client.print() calls, e.g. one for MTA+ then one for L then one for the loco address 123 etc.  It seems that the stream auto-flushes so each print caused a new IP frame to be sent.  Once I realised this I built the command in a buffer and sent it with a single function call.

To change the DCCex code to handle stupid programmers like me seems unnecessary.  The ESP wouldn't provide any help, so the WiThrottle server would have to buffer each client's incoming data until a newline is received, then process the completed command, an unnecessary complication as it works fine already with well-behaved clients!  The commands are all short enough not to be fragmented by the network infrastructure.
The issue I raised is with the WiFi interface module, not the WiThrottle.

FrightRisk, 2021-01-13:
Let's fix the issue and close this one!

Neil-McK, 2021-03-08:
Confirmed added in WifiInterface.cpp in master on 7th March 2021 (#129).

closed 2021-03-08 by Neil-McK