DCC-EX/CommandStation-EX #104: WiThrottle speed erratic

Neil-McK, 2020-12-10:
Another issue I encountered in release 3.0 (and still present now) while using my WiThrottle was that the loco speed changed erratically.  My throttle sends the selected speed to the server in a <V> command, and also honours the incoming <V> responses from the server.  The DCC-ex WiThrottle server is modifying the speed before sending it back.  If I try and reduce the speed by 1, e.g. from 32 to 31, then the server puts it back to 32!

This is evident in WiThrottle.cpp starting at line 265 where the speed is read and then modified by WiTToDCCSpeed(), before it is echoed back to the client.  

The code could be changed as follows to resolve this:
```
// byte locospeed=WiTToDCCSpeed(getInt(aval+1));
byte locospeed=getInt(aval+1);
LOOPLOCOS(throttleChar, cab) {
      // DCC::setThrottle(myLocos[loco].cab, locospeed, DCC::getThrottleDirection(myLocos[loco].cab));
      DCC::setThrottle(myLocos[loco].cab, WiTToDCCSpeed(locospeed), DCC::getThrottleDirection(myLocos[loco].cab)); 
      StringFormatter::send(stream,F("M%cA%c%d<;>V%d\n"), throttleChar, LorS(myLocos[loco].cab), myLocos[loco].cab, locospeed);
}
```
I've tested this, and when I send a speed command the response is now consistent.  Hope this is helpful.

Best regards.

Asbelos, 2020-12-11:
The speed change was added by Steve as i recall.. i have no idea why. It may be something to do with EngineDriver

habazut, 2020-12-11:
I think the observation and fix by Neil is correct. setThrottle takes DCC speedsteps ( 0=stop, 1= emergency stop, 2-127 speeds) and what comes in over WiThrottle has to be converted for setThrottle but _not_ when reporting back. So we set locospeed as the WiThrottle speedsteps and report that back.

To be fixed like this.

Harald.

mstevetodd, 2020-12-11:
@Neil-McK Agreed, we should not be sending the same value to setThrottle() as we return to the client. Thanks for catching that!

I will test and submit a PR shortly.

PR #105 (2020-12-11): FIX: return WiThrottle speedstep, not DCC speedstep, in response to speed change request
Should fix #104
commit b9fdfdd7 (2020-12-11): FIX: return WiThrottle speedstep, not DCC speedstep, in response to speed change request (#105)

* FIX: return WiThrottle speedstep, not DCC speedstep, in response to speed change request.

Should close #104

* use int, not byte for witSpeed
commit 9d92fd94 (2020-12-27): add turnout, sensor and output states to 's'tatus message (#108)

* add support for FireBox_Mk1, reduce heartbeat, separate eStop time

* make match master

* make match master

* Update defines.h

* FIX: return WiThrottle speedstep, not DCC speedstep, in response to speed change request.

Should close #104

* use int, not byte for witSpeed

* add turnout, sensor and output states to 's'tatus message
commit ef95e98a (2021-03-07): FIX: return WiThrottle speedstep, not DCC speedstep, in response to speed change request.

Should close #104
closed 2020-12-11 by FrightRisk