DCC-EX/CommandStation-EX #106: Ethernet shield configuration is not handled correctly.

adanrg, 2020-12-20:
As a user of the original DCC++, I've just found this complete rewrite project and I immediately wanted to try it out. (Great job by the way!)

I'm using an Arduino Mega + Motor Shield + Ethernet Shield combination with statically configured IP address. The production release (v3.0.0) didn't work at all with the ethernet shield, however commit e1ad1f0 on master was a great improvement.

I've found 2 more problems during testing:
1. IP address is always assigned by DHCP, setting the static IP in config.h (IP_ADDRESS) has no effect.
2. Setting the MAC address in config.h (MAC_ADDRESS) has no effect. The one specified in EthernetInterface.h (52:b8:8a:8e:ce:21) is used in all cases.

adanrg, 2020-12-20:
By the way: I also have working / tested fixes for both points on my locally created branch. Let me know, if / how I can contribute here.

habazut, 2020-12-20:
1. The #define IP_ADDRESS in config.h is a leftover from old code. I wonder how many users actually would like to set an IP static in a sketch, but well, we might leave in in there and fix the code even for a minority.

2. I would like to remove the MAC_ADDR from the config.h completely and set it based on the arduino board ID. We would use one of the available prefixes for self-assigned MAC_ADDR  and then take the last octets from the board ID. I'd love to see such code but I don't have an ethernet shield myself so if you could contribute that code, that would be great.

You can contribute will pull requests or just by making your code available somewhere, preferably however in a format that can easily be imported to our git repo.

A 3.0.1 release is upcoming I think.

Thanks,
Harald.

adanrg, 2020-12-20:
My plan was to use a pull request, but for that, I first have to push my branch to GitHub.
This is what I get: (push requires permission)

```
> git push origin adanrg-EthernetShieldFixes
ERROR: Permission to DCC-EX/CommandStation-EX.git denied to adanrg.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

For point 2: Probably the code for user definable MAC address should be kept for now.
I did a quick search for Arduino board ID and the results were mixed. The MEGA16U2 on board used as a "FTDI chip" really has a unique serial number, but that's not readable on the micro-controller side. Is that what you had in your mind as a board ID?

ggee, 2020-12-21:
Create a fork instead and then create a Pull Request from you branch back to master of main rep.

adanrg, 2020-12-21:
Fix avaiilable in pull request https://github.com/DCC-EX/CommandStation-EX/pull/107

adanrg, 2020-12-22:
I've checked the contribution doc, but it doesn't say anything about the PR review process. Should I just request the suggested reviewer?

FrightRisk, 2020-12-27:
That works, but we usually see it. Just having a discussion at the moment. We want to have the MAC set in config.h I think these edits should work and could be merged inter master.

closed 2020-12-27 by FrightRisk