DCC-EX/CommandStation-EX #111: Sensor not working with JRMI over Wifi

markanichols, 2020-12-30:
Firstly, many thanks for your continuing work on DCC++. I have been using the original version successfully  on an Arduino Uno with JRMI on Raspberry PI for some months.

I have been experimenting with your version 3.0.1 with a view to communicating between the Arduino and JMRI over WIFI. I need to continue using JRMI to run an autoshuttle script. I am using an NCE BD20 sensor to detect train position in the script.

DCC++ EX v3.0.1 is installed on an Elegoo Mega with diymore L298P motor shield and Makerfabs Wifi shield. The sensor is defined in the base station and in JRMI, and works correctly when Arduino is connected to Raspberry Pi through USB serial connection. I see correct messages like <q nn> and <Q nn> in JRMI DCC++ monitor when the sensor changes state, and the state changes in JMRI sensor table.

The problem is that I don't see state changes in JMRI when I remove the USB connection and change DCC ++ connection in JMRI to ethernet. Throttles and turnouts work perfectly in this configuration, but sensor state remains 'Unknown' and I see no messages from the sensor in JMRI DCC++ monitor.

Is this expected behaviour or a bug?

Thanks

markanichols, 2020-12-30:
Sorry, the format of the messages I mention has been corrupted because they conatin chevrons. It should read '<q nn>' and 'Q nn' where nn is the sensor id

markanichols, 2020-12-30:
I should mention that the sensor is connected to pins 4 and GND on the Arduino

habazut, 2020-12-30:
The spontanous sensor change messages go today only out on USB as that is the only serial stream which is there from startuptime. Today we do not have any method to loop over all connected streams and so to say broadcast this info to all which might or might not want it. So with the current code that's expected but maybe not desired.

Harald.

Asbelos, 2020-12-30:
Please be aware that the wifi interface is pushing the limits of what the arduino can handle as it is hugely more complex and inefficient thsn a Usb connection. 
The wifi was intended for supporting WiThrottle devices on simple layouts which can be run without the need for a PC running JMRI. 

There are a number of other undesirable issues if yiu try and run JMRI over the wifi.... i strongly recommend you invest in a longer USB cable and avoid trying to do jmri over wifi..

markanichols, 2020-12-31:
Thanks for your replies. I did wonder whether it is an inbuilt limitation. Also I can see that communicating with JMRI over Wifi isn't ideal. In fact my layout is a very small end-to-end with 7 turnouts, and I'm attracted by the smaller base-station format DCC++ EX offers. I mainly need JMRI for this small bit of automation, and probably programming and store loco configurations, about 6 in my case. For the moment, I will live with the USB connection for these purposes. I will also look further at the code here to see if I can contribute anything. Thanks again.

markanichols, 2020-12-31:
I perhaps closed this prematurely. I see now that DCCEXParser.cpp was amended 4 days ago to send all turnout and sensor states when &lt;s&gt; command is received. WIth this version installed I do see the sensor status updated over Wifi when I send that command, and at intervals because JMRI also issues that command regularly when connected over Wifi (but not over USB). Not regularly enough for my script, and I hesitate to poll DCC with the command, but it does perhaps offer something. I will be interested to see what the automation features in development offer.

markanichols, 2021-01-01:
I've had a look at the TPL branch and it looks like the automation features will meet my very simple requirements, and allow me to remove JMRI, perhaps apart from for programming locos. So I will close this issue. If you are looking for testers, I would be interested. I'm a retired IT professional so do have some suitable experience.

closed 2021-01-01 by markanichols