DCC-EX/CommandStation-EX #114: Automatically Issue commands at startup

Asbelos, 2021-01-07:
By being able to add a list of commands to be executed at command station startup, we immediately provide a simple solution to a multitude of existing difficulties with commands that don't "stick" over a power cycle. 

For example:
 Using + commands to configure Wifi
 Overriding waveform/ack  tuning parameters with the D command
 Starting with some diagnostics enabled
 Defining accessories without the assistance of JMRI
 Setting power state at startup
 
 The code for this is simple, consumes no RAM, and can be managed by the automatic detection of an optional  mySetup.h file which, like config.h, will not be overridden on a code update. containing just commands like:
 
  SETUP("D CMD ON")
  SETUP("D ACK ON")
  SETUP("T 1 123 45")

Asbelos, 2021-01-07:
Code implementation suggestion:
a) create DCCEXParser::parse(__FlashStringHelper * cmd)  Copies string to buffer and calls normal parse as if came from serial
b) In .ino setup()  
       #if __has_include ( "mySetup.h")
        #define SETUP(cmd) serialParser::parse(F(cmd)), 
        #include "mySetup.h"
        #undef SETUP
Note: this also allows for the injection of command filters and other custom code  
without having to break open the .ino

Asbelos, 2021-01-08:
StartupCommands branch available for testing

FrightRisk, 2021-01-13:
Would this be able to change the outputs method now that uses iFlags? We need to look in JMRI and see if that is implemented or if it is just in the CS.

Asbelos, 2021-01-14:
Because it can execute any command its completely independent of any particular feature.

Asbelos, 2021-01-21:
Merged in 3.0.3

closed 2021-01-21 by Asbelos