DCC-EX/CommandStation-EX #122: STM32 porting

EasyNetDev, 2021-01-26:
Hi,

I started to do some test with my Nucleo-144 F767ZI board using STM32 MCU. I want to try to port the CS-EX to STM32.
I was able to replicate the old DCC++ PWM overflow IRQ to change the frequency for the PWM output on my STM32 board.
Now I took the code from CS-EX and I'm very confused. The code is very different from DCC++ :).
I found some spots where I have to add my STM32 code, like Timer.h which I have to write it for STM32, but also I've notice some other parts in the code.
There is a documentation for developers where I can find these information?

My idea to port on STM32 is because F767ZI is using a 216MHz MCU with much more EEPROM, RAM and FLASH.
Also other STM32 boards are using 72Mhz MCU and also they are more powerful than Arduino. Also the pinout is compatible with Arduino.

Other advantages: Nucleo F767ZI (and also other STM32 boards) have integrated ethernet + CAN bus protocol (it needs only the CAN transceiver). Maybe in the future I can add LCC protocol integration in the board to be able to have only one board to manage both DCC and LCC through the ethernet  connection.

I would appreciate any help to start the porting.

EasyNetDev, 2021-01-27:
Hi Asbelos,

Thank you for your answer! I will take a look  to that repository.
If you managed to isolate this kind of stuff, it will be much easier for others to read the code. :)

I will give a try to that repository.

Asbelos, 2021-01-27:
For a general discussion on porting find me as UKBloke on our Discord  chat

Asbelos, 2021-02-01:
Just to keep you informed, the nanoEvery2 branch has been updated to use even less device specific code or files. 
If you have to do any more changes, please  keep us informed and we will do our best to keep the code portable.

closed 2025-03-08 by peteGSX