DCC-EX/CommandStation-EX #123: WifiInterface.cpp: password not handled properly in Access Point mode.

Neil-McK, 2021-01-26:
In WifiInterface.cpp, there is an option for the access point to use the password configured in Config.h.  However, this is stored in PROGMEM space, so when it is used in a StringFormatter::send() call it should be specified in the format string as %S (not %s) for it to be handled correctly.
`255  // password configured by user`
`256  StringFormatter::send(wifiStream, F("AT+CWSAP=\"DCCEX_%s\",\"%s\",1,4\r\n"), macTail, password);`
should be
`255  // password configured by user`
`256  StringFormatter::send(wifiStream, F("AT+CWSAP=\"DCCEX_%s\",\"%S\",1,4\r\n"), macTail, password);`

Also, this option has been missed out altogether in the `else` branch when `oldCmd` is false and should be added.

habazut, 2021-01-28:
Have to look at that, you're right, that's probably broken.
Thanks for spotting that,
Harald.
(Strange, I thought I had tested that)

Neil-McK, 2021-03-08:
Confirmed to be fixed in master on 7th March 2021 (#129).