DCC-EX/CommandStation-EX #124: Latest code from last release fails to compile in Arduino Desktop

mthenhaus, 2021-01-30:
In file included from /Applications/Arduino.app/Contents/Java/libraries/Ethernet/src/Dns.cpp:8:0:
/Applications/Arduino.app/Contents/Java/libraries/Ethernet/src/Dns.cpp: In member function 'uint16_t DNSClient::BuildRequest(const char*)':
/Applications/Arduino.app/Contents/Java/libraries/Ethernet/src/utility/w5100.h:457:25: warning: result of '(256 << 8)' requires 18 bits to represent, but 'int' only has 16 bits [-Wshift-overflow=]
 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )
                      ~~~^~~
/Applications/Arduino.app/Contents/Java/libraries/Ethernet/src/Dns.cpp:164:18: note: in expansion of macro 'htons'
  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY | RECURSION_DESIRED_FLAG);
                  ^~~~~
Sketch uses 36118 bytes (14%) of program storage space. Maximum is 253952 bytes.
Global variables use 1811 bytes (22%) of dynamic memory, leaving 6381 bytes for local variables. Maximum is 8192 bytes.

habazut, 2021-01-30:
Well, not much we can do when the libraries ignore issues:

https://github.com/arduino-libraries/Ethernet/issues/151

That was not yesterday I reported that.
Harald.

mthenhaus, 2021-01-30:
Ok, it is an Arduino issue, not DCC-EX, closing.

closed 2021-01-30 by mthenhaus

habazut, 2021-01-31:
Yes, that's correct. So join my nagging in the issue #151 of the library code.

My tests indicate that 
#define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )
can be rewritten as
#define htons(x) ( ((( x )&0xFF)<<8) | ((( x )&0xFF00)>>8) )
and that would make the compiler shut up. You are welcome to test, I haven't even an Ethernet card, so for me it's a bit like trying new swim styles on land.

Alternatively one could use a pragma to tell the compiler that this is intentional but if one can produce code that does the same thing for the same cost without warnings that's better.

(gdb) print /x  ( ((( 0xBEEF )&0xFF)<<8) | ((( 0xBEEF )&0xFF00)>>8) )
$3 = 0xefbe
(gdb) print /x ( ((( 0xBEEF )<<8)&0xFF00) | ((( 0xBEEF )>>8)&0xFF) )
$4 = 0xefbe

Regards,
Harald.