DCC-EX/CommandStation-EX #135: RCN-212 DCC  Functions From F29 to F68

jakespeed1311, 2021-03-11:
Enhancement
It wold be a full Support for DCC to expand the Command Set To all Functions that are defined here RCN-212.
Some Decoder can use that .

Asbelos, 2021-03-12:
Thanks to google translate...
Decoders that support functions 29 to 68 must also be able to control these functions via binary states 29 to 68 Binary state control command, short form This command is two bytes long and has the format: DLLL-LLLL It allows one of 127 binary states to be controlled within the decoder . Bits 0-6 of the second command byte (xlll-llll) must define the number of the binary state, starting with 1 and ending with 127. Bit 7 (D) defines the binary state, with a "1" switched on and a "0" switched off means. The value 0 (xlll-llll = x) is reserved as a broadcast to delete or set the binary states 29 to 127. A command switches the binary states 29 to 127 off ("off"). For RailCom applications, the binary states (XF commands) are defined with the following function numbers LLL-LLLL: 1 Search 2 Fill up with diesel 3 Coal 4 Draw water 5 Fill sand 6 Charge the battery 7-14 reserved 15 General filling command reserved free The function space from 1 up to 28 may only be used for special applications such as RailCom. For general switching functions, only binary state control commands from 29 inclusive may be used. Association of Manufacturers of Digital Model Railway Products Page 8 of 13

Asbelos, 2021-03-12:
Credit https://github.com/YuriyGorvitovskiy/ArduinoDccLibrary/blob/master/DccStandard.h

Feature Expansion Instruction (110)  
	-----------------------------------
	
	The instructions in this group provide for support of additional features within decoder. (See TN-3-05) 
	
	The format of two byte instructions in this group is: 	110CCCCC   0   DDDDDDDD 
			
	The format of three byte instructions in this group is: 110CCCCC   0   DDDDDDDD   0   DDDDDDDD 
	
270 The 5-bit sub-instruction CCCCC allows for 32 separate Feature Expansion Sub-instructions.  
		CCCCC = 00000:  Binary State Control Instruction long form – Sub instruction "00000" is a three byte instruction 
						and provides for control of one of 32767 binary states within the decoder.  The two bytes following this instruction 
						byte have the format DLLLLLLL   0   HHHHHHHH".  Bits 0-6 of the first data byte (LLLLLLL) shall define the 
275  					low order bits of the binary state address; bits 0-7 of the second data byte (HHHHHHHH) shall define the high 
						order bits of binary state address.  The addresses range from 1 to 32767.  Bit 7 of the second byte (D) defines the 
						binary state.  A value of "1" shall indicate that the binary state is "on" while a value of "0" shall indicate that the 
						binary state is "off".  The value of 0 for the address is reserved as broadcast to clear or set all 32767 binary states.  
						An instruction "11000000   0   00000000   0   00000000" sets all 32767 binary states to off. 
280 
						Binary states accessed with all high address bits set to zero would be the same as accessed by the short form of the 
						binary state control.  Command stations shall use the short form in this case, i.e. Binary State Controls 1 to 127 
						should always be addressed using the short form.  Decoders supporting the long form shall support the short form as 
						well. 
*/
#define DCC_MF_KIND8_LONG_STATE_CONTROL 		(0xC0)

/**						
285 
		CCCCC = 11101:  Binary State Control Instruction short form – Sub-instruction “11101” is a two byte instruction 
						and provides for control of one of 127 binary states within the decoder.  The single byte following this instruction 
						byte has the format DLLLLLLL.  Bits 0-6 of the second byte (LLLLLLL) shall define the number of the binary 
						state starting with 1 and ending with 127.  Bit 7 (D) defines the binary state.  A value of "1" shall indicate the binary 
290  					state is "on" while a value of "0" shall indicate the binary state is "off".  The value of 0 for LLLLLLL is reserved as 
						broadcast to clear or set all 127 binary states accessible by the short form of the binary state control.  An instruction 
						"11011101   0   00000000" sets all 127 binary states accessed by this instruction to off. 
						
						Binary State Controls are quite similar to Functions, as they may control any output, sound or any other feature of 
295  					digital nature within a decoder in direct response to a packet received.  But Binary State Controls do have a different 
						access method and function space.  Therefore they have a different name. 
						
						Binary state control packets – both short and long form – will not be refreshed.  Therefore non-volatile storage of 
						the function status is suggested.  When operations mode acknowledgment is enabled, receipt of a Binary State 
300  					Control packet shall be acknowledged accordingly with an operations mode acknowledgment.  Consult the 
						Technical Note(s) for additional information on this instruction.  (See TN-4-05)

Asbelos, 2021-03-12:
IMHO this appears simple to include short and long binary, especially as these do not need reminders.

Asbelos, 2021-03-16:
Jake, please can you take a look at the assortedBits branch where I have implemented the short and long binary state commands but have no way of knowing if they work.

jakespeed1311, 2021-03-20:
HI,
tanks, yes will do this till 26.3. i am not at home.

jakespeed1311, 2021-04-05:
Hi, to put it better I meant the function commands. F13-F68
I tried to send them with this code but it does not work.
  
2.3.4 Function control F13-F68
These commands are two bytes long and have the following formats:

Function control F13-F20: 1101-1110 DDDD-DDDD
Function control F21-F28: 1101-1111 DDDD-DDDD
Function control F29-F36: 1101-1000 DDDD-DDD
Function control F37-F44: 1101-1001 DDDD-DDDD
Function control F45-F52: 1101-1010 DDDD-DDDD
Function control F53-F60: 1101-1011 DDDD-DDDD
Function control F61-F68: 1101-1100 DDDD-DDDD

I tried to send it directly through the schedulePackert but it is not happening. 
Strangely it works with deb code b[nB++] = { B11011000}; for F29-F36.

`setFunctions37to44(uint16_t address, uint8_t functions)	//F37 F38 F29 F40 F41 F42 F43 F44`
`{`
`if (address == 0)	//check if Adr is ok?`
`return false;`
`functions = B10101010;   //170`
`uint8_t b[4];`
`uint8_t nB = 0;`
` if (address > 127)`
 `   b[nB++] = highByte(address) | 0xC0;    // convert train number into a two-byte address`
` b[nB++] = lowByte(address);`
`	             `
`b[nB++] = { B11011001}; `
`b[nB++]`  = functions;  //significant functions (F37--F44)`

`DCCWaveform::mainTrack.schedulePacket(b,` nB, 3);
``

Is there anything else to consider ?
greetings
Jake

habazut, 2021-04-08:
What does not work in which way?

We send reminders up to F28 (RCN-212 2.3.4) F29 and above we send 2,3.5 commands and from F128 we send 2.3.6 style commands. According to the standard, if your decoder understands 2.3.4 command F29 and above it should as well understand 2.3.5 commands. See last sentence i section 2.3.4. "Decoder, die die Funktionen 29 bis 68 unterstützen, müssen diese Funktionen auch über die Binärzustände 29 bis 68 steuern können."  So with what Decoder are you trying to do something and which command set works and which does not?

Greetings,
Harald.

jakespeed1311, 2021-04-09:
Hi,
first the 2.3.5 and 2.3.6 Binary Commands are working.
I had an Isue with my DCC Monitor that is fixed now. Also i have an ESU 4 that works with that Binary Sequences. 
But i am Using the 2.3.4 Commands to set F29 to F68. I try my tests with the malfuction DCC Monior and i asked about if i missed something with the Wafeform generator input. I try to understand the DCCpp code. 
But now  it works, and i have tested all ok. Any yes the Binary Comand have more charm.
I think there are no reminders need.
Tanks 
Jake
@Asbelos  Tanks for Binary Commands

closed 2021-04-09 by jakespeed1311