DCC-EX/CommandStation-EX #136: Support DCC with Speed steps 14 and 28 for old Decoder

jakespeed1311, 2021-03-11:
enhancement
Command extension to the decoder travel error for 14 /28 speed steps. Although current decoders have 128 speed steps there are still older models with old to very old decoders.
Since our model railroaders are over 40, it is not uncommon to operate certain locomotives with 14 or 28 speed steps.
Definition also in  RCA-212  2.2.1

It would support DCC-ex very much to support older driving decoders.
Tanks
 Jake

habazut, 2021-03-16:
I have now created a branch called 28step which enables to switch the whole CS to 28 speedstep mode with the <D SPEED28> command (and back with <D SPEED128>). Please test. I am not so happy to implement 14 step because it's a pain to dig out the value of F0 and put that into the same packet as the 14 speedsteps. So we kinda voted against 14. But here you have 28. Be aware that this is a compromise which will switch the _whole_ CS to 28step mode but that way no changes need to be made in the DCC++EX protocol.

Harald.

Asbelos, 2021-03-16:
Unfortunately the commands don't show in github or the email due to the < > chevrons. . Let's try markup... 
They are  ```< D SPEED28> ```   and  ```<D SPEED128 >```

FrightRisk, 2021-03-17:
The only thing is that these are not really diagnostic commands. It is "settings". Do we need to make a distinction? We reserved U for User commands but I think that was for things like the filter capability where WE never use it for anything. Even if we did want to avoid D, then it would get lost in its own category instead of lumped in with Diags

habazut, 2021-03-17:
I hate when editors are not WYSIWYG (What You See Is What You Get). If I type a greater than here I want to see on in the email....
Then on the actual "D" command: I added it as "D" because that was easy for test. If we want a "user settings" command (which we probably want) we should create one in the next release. But then we should commit to everything in the user settings to be supported now and in the future so more thinking than for a "D" command needed before release.

Harald.

jakespeed1311, 2021-03-17:
HI,
ok i thinked about to expand the speed command with one Number. If the number is not here it is like the old command.
Impenting like the <1> comand  it switches with the sum ob parameters given.
So we do not need an new command, but yes an change on it. But ist was working as before without it.
And D would be for debug.
greedings
Jake

habazut, 2021-03-17:
So does the code work for you as is.

Currently there is no place in the internal table of decoders to remember which decoder has how many speed steps and I would like to keep it that way if possible. The problem will get smaller for every year.

A protocol change has to be followed up by JMRI, Win-Digipet, Rocrail and all the others which support DCC++EX, at least if they want to use this as well. 

Then I don't like the t command with it's useless extra parameter anyway and we have to decide what to do about that in the future. 

Harald.

FrightRisk, 2021-03-18:
I don't understand Jake's comments. When he says "But it was working as before without it", is he saying that it worked with Classic? Or is he saying that the command would work the same if the flag was not set for 28 steps? To reword it, does he mean, "But it would work as before without it (if the flag was not set)"? In any case, I think we have done enough with this feature. It is a good compromise. We can't be all things to all people and have to focus on our mission which is moving forward with technology and looking to the future.

Asbelos, 2021-03-18:
I think Jake was suggesting an update to the throttle command but that is definitely NOT a good idea.

jakespeed1311, 2021-03-18:
Hello,
So first I find DCCex very good, unfortunately I also do not fully understand the C code. I also like to cut off the old braids would it only be so simple.
To make it clear I also think 128 speed levels would be great, good if all had that and then also used. 
Unfortunately, the reality is still quite different. Despite regulations to use 128 speed steps, there are still old decoders on the road with 28 speed steps.  And then to say you can not drive because ... then hangs the same old control center on the track that supports it.  I have experienced myself that are still valid and it moves yet ...   Yes may be that it thins out, but I think they will still be there for 7-10 years.   May the solder joints of these decoders age rapidly, fun.

The memory consumption with the byte per locomotive the Fährt I have not thought, but I also have a Teensy 4 on the run. (Thanks Mike). So yes that can be that a Nano gets there problems.
The LCD commands are also everywhere in the code without #If defined (LCD) e.g. to comment them out. May be that the current solution is much better (Ram consumption), so far that understand. Well, I can't program that well, so my opinion is totally wrong.
Maybe there are other reasons e.g. the current speed controllers don't support it and so on.

Based on the current description I found the idea just ok.
 

< = Start delimiter of a DCC++ EX command. (A space after < is not required but acceptable)
t = (lower case t) This command is for a decoder installed in an engine or simply a "cab".
1 = deprecated. We no longer use this but need something here for compatibility with legacy systems. Enter any single digit.
03 = CAB: the short (1-127) or long (128-10293) address of the engine decoder (this has to be already programmed in the decoder) 
         See Programming Commands bellow.
20 = SPEED: throttle speed from 0-126, or -1 for emergency stop (resets SPEED to 0)
1 = DIRECTION: 1=forward, 0=reverse. Setting direction when speed=0 or speed=-1 only effects directionality of cab lighting for a 
      stopped train.
> = I am the end of this command

Sorry for misformating this.

And there is a number after the <t which should be free, and is only there to be compatible with DCCpp. 

But now an example:
If I now want 28 speed steps for this loco, I append the " X". W here which is meant as a placeholder. 
"<T 1 20 1 X>"  
X =
0 = 128 steps but not needed
1 = 28 Steps must be set.
2 = 14   

If the X is missing and is a > it is the normal command. It works then just like now with 128 steps.  And it could be recognized relatively well with the parameter number of the command. As I have seen here with other commands also.
You might not need to save the steps when the command with extension comes simply the old DCC command for 14/28 steps sent. 

But this is now just my thought been.  To DCCex it now simply better to the now me known practice to adapt, without large expenditure. As I thought...
Please see it as a brainstorming, even if I have not yet internalized your concept.
Thanks anyway that you deal with the proposal.
Thanks
Jake

Translated with www.DeepL.com/Translator (free version)

Asbelos, 2021-03-18:
Jake, 
When you set the CommandStation into 28 speed step.mode, all your 128 step locos will still work , all the known throttles including withrottle versions will still work for 28 and 128 step locos.

If we change the command to include the extra parameter, NONE of the existing throttles will be able to drive your 28 step locos. You will have to write your own throttle.

The beauty of open source is that you are free to do that.

jakespeed1311, 2021-03-20:
Hi,
are you shure that with my change all Other trottles not running? I don't think so, if the now working  command was parsed it ignores the parameter witch can appended. 
And also i do not want to swich the complete ComandStation into 28, only one Loco.
But i will try to change the command and test by myself whats happending wit JMRI or the wifi trottle. 
If i get you right that can't work with that change, i wilfl see it an maybe understand wy it is not working.

Asbelos, 2021-03-20:
With your change, suggestion  all existing throttles will just continue to work at 128 and will not be able to drive a 28 loco.
With our change, ALL existing throttles will be able to drive ALL your locos... because all 128 step locos work OK in 28 steps. 

Changing throttles to use your additional command parameter is hundreds of lines of code in at least 5 different languages. You're welcome to try.

jakespeed1311, 2021-03-21:
hi,
Yes thats true, if the Trottele do not send the expanded Command it runs with 128.
I thought to control by DCC addresse whether with 128 or 28/14 we drove.

ok,now i get you now! I will test  the 28step branch next weak.

To be clear how it works now.
I have to send the  <D Speed28> to bring the Command station to Global28 Speedstep mode. 
And i can drive an 128 and an 28 Speed stepp loko to the same time on the Main ?
Or can i drive ony 28 Or 128 Speedstep  same time ?

I think gloabl means 28 OR 128 in the same time on Main ? 
Right?

Asbelos, 2021-03-21:
When you set the command stationto 28 speed steps, it will send a different kind of DCC speed message to all the locos. 

Your 128 locos WILL understand this message and work normally. Most people will not see any difference. 

So yes, if you set 28 step mode, you can drive 28 and 128 locos at the same time.

jakespeed1311, 2021-04-08:
Hi,
I testet this Branch now, ok late but it is working as espected
Tanks.
Jake

closed 2021-04-08 by jakespeed1311