DCC-EX/CommandStation-EX #151: Display: Move IP address and port to the same line

FrightRisk, 2021-05-18:
[image]

FrightRisk, 2021-05-18:
in EthernetInterface.cpp

  //LCD(4,F("IP:%d.%d.%d.%d"), ip[0], ip[1], ip[2], ip[3]);
  //LCD(5,F("Port:%d"), IP_PORT);

  LCD(3, F("%d.%d.%d.%d:%d"), ip[0], ip[1], ip[2], ip[3], IP_PORT);

Neil-McK, 2021-05-19:
Bear in mind that the worst case the up address may be 15 characters and
the port 5 characters, that's 21 characters total.  In general it would
probably be xxx.xxx.x.xxx:xxxx with defaults, but this is still 18
characters, too long for a 1602 LCD.  May need to be conditional on screen
size.

tz3p9v, 2021-05-20:
The compile time test of MAX_MSG_SIZE does check for size of display as MAX_MSG_SIZE = 20 is defined in config.h as the size of the display -- true max IP and Port could be 21.  So test could be added to ensure display length < MAX_MSG_SIZE, but would need to be run-time.  As typical display is 18 as shown in picture of my display, I suggest no need for run time test.  User can correct by removing MAX_MSG_SIZE from config.h, which will default to 16 and a two line display.

FrightRisk, 2021-05-21:
There's always a caveat ;) What do you think Neil?

closed 2025-03-08 by peteGSX