DCC-EX/CommandStation-EX #162: OutputData struct in Outputs.h ID should be int.

FrightRisk, 2021-05-24:
It is currently uint8_t. It should be int to match the rest of the code. This limits Outputs id range to 255.

Neil-McK, 2021-05-24:
The HAL work under way has revised the output ID to an int (valid range 0-32767).  This will hopefully be incorporated into the master branch in due course.  Changing the structure element sizes will cause existing EEPROM contents to be mis-interpreted with unpredictable consequences that are likely to draw complaints.  I would therefore advise against changing this as a bug fix.

FrightRisk, 2021-05-24:
I am thinking not a ton of people are using this, and if they are and we tell them the data structure is different, they can choose to update or not. How do you feel about a breaking change to really make EEPROM work in the best way it can? Haba had a valid question as you allude to, is there a case for outputs over 255? What kind of layout would need that? A mega has the most pins and only has around 50. Then again, I don't want to be shortsighted like the NMRA was with Function buttons. Maybe 2 years from now there will be a board with 1024 output pins.

closed 2025-03-08 by peteGSX