DCC-EX/CommandStation-EX #164: COMMON FAULT PIN ACTIVE

rmccown, 2021-05-29:
I am trying to load DCC++ EX onto my Arduino Uno / Pololu 33926 motor shield that I am currently running DCC++ on.  DCC++ runs correctly on this hardware.  I cloned the DCC++EX repo, copied and changed the config for the Pololu board, compiled and uploaded it to the Uno.  I have made no other changes.

When I send the start command, I get repeating messages "COMMON FAULT PIN ACTIVE".  

This is the complete log from startup, to the point where the messages repeat

```
<* LCD0:DCC++ EX v3.1.0 *>
<* LCD1:Starting *>
<* MotorDriver currentPin=A1, senseOffset=0, rawCurentTripValue(relative to offset)=166 *>
<* MotorDriver currentPin=A0, senseOffset=0, rawCurentTripValue(relative to offset)=166 *>
<iDCC-EX V-3.1.0 / UNO / POLOLU_MOTOR_SHIELD G-90487d2>
<* Signal pin config: normal accuracy waveform *>
<* LCD1:Ready *>
<* LCD2:Free RAM= 904b *>
<* LCD2:Free RAM= 867b *>
<p1>
<* *** COMMON FAULT PIN ACTIVE - TOGGLED POWER on MAIN *** *>
<* *** COMMON FAULT PIN ACTIVE - TOGGLED POWER on PROG *** *>
<* *** COMMON FAULT PIN ACTIVE - TOGGLED POWER on MAIN *** *>
<* *** COMMON FAULT PIN ACTIVE - TOGGLED POWER on PROG *** *>

```

There is no output to the main or prog tracks.  Re-loading DCC++ onto the hardware reverts me back to a working system.

Asbelos, 2021-05-29:
Did you remove the jumpers when running dcc-ex?

habazut, 2021-05-29:
Do you have an unmodified Pololu shield or did you modify the shield as Gregg did for DCC++ (Classic)? In the second case the pin definitions do probably not match.

So what have you as fault pin and what is the voltage on it? Does this happen only when you have layout connected or even with all output from the shield disconnected?

The internal short current detection on the motor driver chip is very trigger happy.

Regards,
Harald.

rmccown, 2021-05-29:
> Did you remove the jumpers when running dcc-ex?

I have tried both with and without the jumpers, and get the same result.

> Do you have an unmodified Pololu shield or did you modify the shield as Gregg did for DCC++ (Classic)? In the second case the pin definitions do probably not match.

It has the three traces cut as defined in https://github.com/DccPlusPlus/Documentation/blob/master/Motor%20Shield%20Pin%20Mappings.pdf

I could re-jumper them, but that doesnt seem to be a solution to a "pin active" type of warning.

habazut, 2021-05-29:
The definition in DCC++EX is for a non modified board.

You need at least reconnect the pin 12 cut, you can set a header for a jumper there.

It unconnected, pin 12 floats and can have any value.

DCC++ Classic did not take care of pin 12 at all which meant that the board could block without warning unless you manually toggle it off and on again.

 I think you can as well replace the -4 with UNUSED_PIN as you have conneted that to 9.

Harald.

rmccown, 2021-05-29:
Thanks Harald, I'll get soldering in some jumpers and see what happens.

rmccown, 2021-05-29:
And, that was it.  Works now.  We now return you to your regular program.

closed 2021-05-29 by rmccown