DCC-EX/CommandStation-EX #167: Current detect offset doesn't work when track power is off

FrightRisk, 2021-06-07:
From a email support request: 

The "self initializing function doesn't work properly when the track isn't switched on...
Without the DCC signal (or at least the enable signal) the motor H bridge is switched off. So there is no current sense output signal available...
I hope you understand what I mean..

I tried to extend the declaration string in the config.h file with testing values 555 and 666 for manual overriding the self measured value

 config.h:
```
#define MY_MOTOR_SHIELD F("MY_MOTOR_SHIELD"),\
   new MotorDriver(11, 13, UNUSED_PIN, UNUSED_PIN, A1, 9.766, 6000, 7, 555), \
   new MotorDriver(3, 12, UNUSED_PIN, UNUSED_PIN, A0, 2.99, 2000, UNUSED_PIN, 666)
```

and the MotorDriver.h
```
MotorDriver(byte power_pin, byte signal_pin, byte signal_pin2, int8_t brake_pin,
           byte current_pin, float senseFactor, unsigned int tripMilliamps, byte faultPin, int currentOffsetOWN);

```
and the MotorDriver.cpp
```
MotorDriver::MotorDriver(byte power_pin, byte signal_pin, byte signal_pin2, int8_t brake_pin,
                         byte current_pin, float sense_factor, unsigned int trip_milliamps, byte fault_pin, int currentOffset_OWN) {
```
and made a hard override...
```
    senseOffset=analogRead(currentPin); // value of sensor at zero current
    senseOffset=currentOffset_OWN;

```
As far as I can see, this seems to work... but I am not sure and hoping not to have broken something else...

Asbelos, 2021-06-07:
I guess that depends on the shield in use because Ash found it worked for him. We obviously can't measure the offset with the power on ... if you want to add this value to the end of the motor shield constructor then it needs to be optional to prevent breaking existing code and to have a default value that can be recognised as "please do what 3.1 does and test it"

Ash-4, 2021-06-07:
I have not had an issue with the offset.  The external current sensors I used show the correct offset, even with track power supply unplugged; the current sense is powered from the 5V pin on the Arduino.  (Also a zero offset on the L298N -- no voltage drop over the 1.5 ohm resistor with no power.)  I was not aware that a motor shield would not be a zero offset.

I wonder if this is like the issue we found that led to connecting 5V to the motor board, in addition to the enable and signal pins.  You might recall how our wiring is different from the transistor-inverter circuit previously used; not only can we use two signal pins, but we also wire the motor board VCC to Arduino 5V pin.
https://dcc-ex.com/advanced-setup/supported-motorboards/IBT_2-motor-board-setup.html

It would be good to see some diagnostics, components, wiring...  I'm ok with a parameter for offset.  But if the reason it is needed is due to a missing 5V connection, I would still encourage the 5V connection -- and perhaps the 5V would result in better precision when reading CVs.

FrightRisk, 2021-06-07:
Can you find that thread for me on Discord @Ash-4 ?

Ash-4, 2021-06-07:
motor-shields from 3/8/2021:  https://discord.com/channels/713189617066836079/734955580149727243/818552739348217885  

. . .   The PWM signals actually continue to run even if EN is off. The PWM signals will slowly charge the VCC cap (C2) thought the ESD structures in U1.  And then "interesting things" start to happen with odd signal levels. The solution is to connect the IBT_2-VCC to the Arduino 5V and then the EN will actually perform    . . .

MerijnD, 2021-06-09:
Hi all,
First off all congratulations with the 3.1  release! Great Job!

This question was coming from me. Thank you FrightRisk for publishing it to this forum!
I am using  the "BTS7960 High Current 43A H-Bridge Motor Driver" probably a cheap clone...

When connecting the Vcc and the ground (5V) and the power supply (24V) 
IS (current sense) output pins stay low in millivolts. 
When the enable pin is set high, the IS out pin makes 1,5V 
Both situations without the track connected, so no current can flow

Connecting a dummy load from 0,5A up till 10A, brings the IS pin 0,5V per 1A higher (linear)
This is all unrelated to the PWM signal 

Everything else works fine!
But because during initialising the current offset is determent with the enable signal low, the IS offset is not measured correctly in my case… Therefore I override it with a hard value… (only hoping not to damage other parts of code)

Only for your interest... 24V@10A is because I have LGB G scale

Best regards Merijn

Ash-4, 2021-06-10:
I am unable to duplicate the issue.
  
This seems to be the IBT_2 board in the documentation.  Still missing the confirmation that wiring and motor board definition are correct/consistent, and what board/shield is being used for programming track.  Does wiring match the diagram or the fritzing?  If so, motor board definition should show two signal pins.  (and I don't understand the fault pin 7.)

These lines are from the serial monitor.  Same results with track power supply connected using 15V or 25V power.  Tracks not connected.

<* LCD0:DCC++ EX v3.1.0 *>
<* LCD1:Starting *>
<* MotorDriver currentPin=A1, senseOffset=0, rawCurentTripValue(relative to offset)=409 *>
<* MotorDriver currentPin=A5, senseOffset=0, rawCurentTripValue(relative to offset)=600 *>

<p1>
<c CurrentMAIN 30 C Milli 0 6000 1 6000>
<a 5>

my current factors are 4.88 for L298N, and 10.0 for IBT_2.  Trip currents at 2000 and 6000.

I have not added the diode protection circuit -- just the 10k resistor shown in fritzing.  

If there is a motor shield installed, IBT_2 current sense would not normally use pin A0 or A1.
BTS7960 status pin 'IS' is used as a combined current sense and error flag output (see data sheet).  Could the 1.5V reflect incorrect wiring from the power supply or Arduino?

Is it a problem if the current offset is used to compensate for anomalous/unexpected results?

MerijnD, 2021-06-19:
Hi Ash-4. Thanks for helping me out.

You are Right, I am using the IBT_2 board. And no programming track or second shield. Programming is done via another decoder at the workbench.
The wiring is done via a NPN transistor to invert the signal to make A and !A. (the old original DCC project didn’t support 2 outputs on the MEGA. And this is working fine. No problems controlling the trains. ( this solution is also know on tutorials at the www)

The fault pin was included for testing because I was looking for a way that when a overload was detected, the timeout function was disabled. ( I don’t want the auto reclosure function because of the higher current). Sadly it didn’t work , fault pin also starts a timeout instead of switching off… So please ignore this setting.

The “senseOffset=0” in your serial monitor output “is the problem”… This value is 0 because this is measured by the MEGA while the enable signal is off during the setup routine. This puts the IBT_2 in standby mode. As soon as the track is switched on (enable signal is high), the IBT_2 board becomes active and generate a DC offset without any current (Even with the tracks disconnected)
As also found in the datasheet, the BTS7960 has a huge possible error at his IS pin (current sense output ). This is the cause of the problem.

Ash-4, 2021-06-20:
I had been using pins 3,4,7 jumpered on IBT_2 (and similar for L298N) for the transistor/inverter circuitry, such as this diagram shows:  https://www.trainboard.com/highball/index.php?threads/esp32-command-station.112634/page-13#post-1047918
or http://sourpuss.net/projects/trains/misc/DCCppESP32.png

But we found that IBT_2 pin 7 should be connected to the +5v from Arduino -- not powered on/off by the enable signal.  Only connect enable to pins 3 and 4 of IBT_2.   Connecting pin 7 on the IBT_2 to +5v (and not to pins 3 and 4) should ensure that the IBT_2 logic is powered at all times that the Arduino is.

Could you give this a try and see if it resolves the problem?
I'm also curious on the offset you expect to see.

Asbelos, 2022-04-20:
I believe this should be closed.
The enable pin is used to enable the track output... it is NOT there to enable the board logic which should remain powered on at all times.

closed 2022-04-20 by Asbelos