DCC-EX/CommandStation-EX #168: Turn off Auto-Retry after short circuit detect

FrightRisk, 2021-06-07:
This may be the second or third user to ask for this. Merijn D. from the Netherlands mentioned that he has an outdoor garden setup and it uses 10A. He would feel better if it shorted and he had to issue a <D RESET> instead of automatically trying to reconnect power.

Asbelos, 2021-06-07:
If you wish to manage this differently, look at DCCWaveform.cpp and look for 

  case POWERMODE::OVERLOAD

You can remove the power on just below it.

Ash-4, 2021-06-09:
from earlier notes:
     Perhaps something like this:
 
 [include in mySetup.h]
                  <D SHORTMS 100>

If a short (or overcurrent) persists beyond 100ms, tracks are powered off.  And LCD is updated to report the short condition, along with current and time.
 -- Add info on recommended parameters to avoid issues with autoreversers, etc.

Ash-4, 2021-06-09:
-- is the read of CVs related
[overload_44.xls](https://github.com/DCC-EX/CommandStation-EX/files/6621365/overload_44.xls)
:
Using the trip current for programming, NMRA 250ma, I noticed 44 OVERLOAD messages when reading/writing 1900+ CVs.  Changing the 250ma is an option, but I thought that the 250ma would not have been enforced during the CV read/write on the programming track. Here are a few of the 44; quite interesting to see the exponential progression.
12:40:04.080: 	RX: * *** PROG TRACK POWER OVERLOAD current=1415 max=248  offtime=20 *** *
12:40:12.498: 	RX: * *** PROG TRACK POWER OVERLOAD current=1244 max=248  offtime=40 *** *
12:40:16.550: 	RX: * *** PROG TRACK POWER OVERLOAD current=1473 max=248  offtime=80 *** *
12:40:20.315: 	RX: * *** PROG TRACK POWER OVERLOAD current=1322 max=248  offtime=160 *** *
12:40:28.577: 	RX: * *** PROG TRACK POWER OVERLOAD current=1434 max=248  offtime=320 *** *
12:40:32.797: 	RX: * *** PROG TRACK POWER OVERLOAD current=1322 max=248  offtime=640 *** *
12:41:00.049: 	RX: * *** PROG TRACK POWER OVERLOAD current=1415 max=248  offtime=20 *** *
12:41:06.566: 	RX: * *** PROG TRACK POWER OVERLOAD current=1224 max=248  offtime=40 *** *
12:41:06.997: 	RX: * *** PROG TRACK POWER OVERLOAD current=1361 max=248  offtime=80 *** *
12:41:14.577: 	RX: * *** PROG TRACK POWER OVERLOAD current=1390 max=248  offtime=160 *** *
12:41:18.040: 	RX: * *** PROG TRACK POWER OVERLOAD current=1083 max=248  offtime=320 *** *
12:41:20.354: 	RX: * *** PROG TRACK POWER OVERLOAD current=1346 max=248  offtime=640 *** *

closed 2025-03-08 by peteGSX