DCC-EX/CommandStation-EX #180: Upload DCC++ EX to Mega2560 TimerA fault

muddster, 2021-07-20:
I installed DCC++ EX in my Arduino folder and tried to upload it to a Mega2560. I received the following error.

sketch\Timer.cpp.o (symbol from plugin): In function `TimerA':
(.text+0x0): multiple definition of `__vector_20'
sketch\DCCTimer.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Does anyone know why this is happening and how to resolve it?

Many thanks, Brian Mudd

PilotinControl25, 2021-07-20:
Make sure you have all the timer libraries installed....use the library manager

FrightRisk, 2021-07-20:
Hi Brian, you most likely had an older version and copied over it.  It is picking up definitions from a file that is no longer part of the code and shouldn't be in that folder (Timer.cpp). Start clean. Delete the folder and then unzip it from here: https://github.com/DCC-EX/CommandStation-EX/releases. That will make sure you have only the files you need. If you already had a config.h file, it will not be overwritten. If you need one, just rename config.example.h to config.h. Make any changes you like to config.h and upload.

muddster, 2021-07-20:
Thank you very much FrightRisk, that solved the problem.

closed 2021-07-20 by muddster