DCC-EX/CommandStation-EX #198: exInstaller on Linux does not have serial-discovery, only serial-discovery.exe

petr-kalinin, 2022-01-19:
I've just downloaded exInstaller (exInstaller-linux-x64.tar.gz) from "Automated Installer" from https://dcc-ex.com/download/commandstation.html . I start it, it starts writing some logs into the window, and then crashes, outputting the following to the console:

```
petr@kalinin-hp-2020:~/rail/exInstaller-linux-x64$ ./exInstaller
Unhandled exception. Grpc.Core.RpcException: Status(StatusCode="Unknown", Detail="error getting port list from serial-discovery: starting discovery process: fork/exec /home/petr/rail/exInstaller-linux-x64/packages/builtin/tools/serial-discovery/1.2.1/serial-discovery: no such file or directory")
   at Grpc.Net.Client.Internal.HttpClientCallInvoker.BlockingUnaryCall[TRequest,TResponse](Method`2 method, String host, CallOptions options, TRequest request)
   at Grpc.Core.Interceptors.InterceptingCallInvoker.<BlockingUnaryCall>b__3_0[TRequest,TResponse](TRequest req, ClientInterceptorContext`2 ctx)
   at Grpc.Core.ClientBase.ClientBaseConfiguration.ClientBaseConfigurationInterceptor.BlockingUnaryCall[TRequest,TResponse](TRequest request, ClientInterceptorContext`2 context, BlockingUnaryCallContinuation`2 continuation)
   at Grpc.Core.Interceptors.InterceptingCallInvoker.BlockingUnaryCall[TRequest,TResponse](Method`2 method, String host, CallOptions options, TRequest request)
   at Cc.Arduino.Cli.Commands.V1.ArduinoCoreService.ArduinoCoreServiceClient.BoardList(BoardListRequest request, CallOptions options) in C:\Users\dex35\source\repos\exInstaller\exInstaller\obj\Release\netcoreapp5.0\linux-x64\CommandsGrpc.cs:line 1328
   at Cc.Arduino.Cli.Commands.V1.ArduinoCoreService.ArduinoCoreServiceClient.BoardList(BoardListRequest request, Metadata headers, Nullable`1 deadline, CancellationToken cancellationToken) in C:\Users\dex35\source\repos\exInstaller\exInstaller\obj\Release\netcoreapp5.0\linux-x64\CommandsGrpc.cs:line 1318
   at exInstaller.Utils.ArudinoCliHelper.DetectBoard() in C:\Users\dex35\source\repos\exInstaller\exInstaller\Utils\ArudinoCliHelper.cs:line 326
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
Aborted (core dumped)
```

I've checked and indeed there is no file `/home/petr/rail/exInstaller-linux-x64/packages/builtin/tools/serial-discovery/1.2.1/serial-discovery`, only `/home/petr/rail/exInstaller-linux-x64/packages/builtin/tools/serial-discovery/1.2.1/serial-discovery.exe`. The original .tar.gz archive also contains only .exe.

petr-kalinin, 2022-01-19:
I've downloaded `serial-discovery` linux binary from https://github.com/arduino/serial-discovery/releases/tag/v1.3.1, put it to `/home/petr/rail/exInstaller-linux-x64/packages/builtin/tools/serial-discovery/1.2.1/serial-discovery`, after which exInstaller managed to proceed. After that, uploading failed with a similar problem with `packages/builtin/tools/ctags/5.8-arduino11/ctags.exe` (only .exe present), I've downloaded linux binary from https://github.com/arduino/ctags/releases, after which I managed to properly upload code to arduino.

dexslab, 2022-01-19:
Yes this will be fixed in the next release for linux that is the issue we have with it currently and sometimes the installer doesnt download the proper packages it needs which is why we included the folder.