DCC-EX/CommandStation-EX #226: Option for potentiometer and encoder speed control

FrightRisk, 2022-04-20:
I am looking for us to come up with one or two options for a simple attachable speed control that a Conductor could easily connect. We have a lot of discussion on Discord about this, but I am compiling it here since there are so many variables. 

1. Use a potentiometer or an encoder
2. If encoder, what type (linear vs. absolute)
3. What happened to Neil's suggestion for a HAL driver to an encoder?
4. There could be an up and down button type throttle too. (press or press and hold)
5. Need at least 3 buttons (programmable?)
6. What should the code look like?
7. If the control is a pot, what happens if you switch locos? Resistance is absolute, so a new loco could lurch

Harald had some code if he could post it to use a pot to control a loco. We can also use EX-RAIL. Does it have an analog read yet like it does for buttons? Is there some combination of code in the .ino file and an EX-RAIL script that we should use? It seems the simplest (if we need code anyway) is code in the loop of the .ino file to control it all.

Neil and Harald talked about an the encoder HAL that starts at 0 at power on and goes to a pre-defined max. Turn past max and nothing happens. Turn the other way and it immediately goes slower (no absolute zero). Buttons would handle stop and reverse.

We could also use an ADS1115 ADC with I2C code, but that adds complexity, expense and another board.

Regardless, it sure would be fun to test the motorized slider control, first as hardwired and then in a wireless throttle. Here:
https://www.youtube.com/watch?v=r12wG7ebvnY
and code here: https://pastebin.com/bgF5uWBa

The interesting thing about a motorized fader would be feedback from the CS setting the position. One benefit may be that if switching locos, the last speed is remembered and the motor drives the slider to the correct position. How else could a motorized slider be used?

Obviously, this is not a full throttle since it would only have 3 buttons and a knob, but perhaps that could be expanded easily. The idea is to have something that does not require another processor and sketch, it could all be done with 3 or 4 buttons or a keypad, and a pot or encoder and a version of the CS users could download.

alex-code, 2022-04-20:
Most encoders have a push button so that could be used too.

My couple Q's
How would you select a loco? 
Would that require an LCD or some display?

Something like this, an MCP23008 (16 pin version available too) could be used to keep it all i2c to limit wires and make it easier to wire up.

FrightRisk, 2022-04-21:
In its simplest form, it could control ONE loco, maybe to a hard-coded address, like "3". We could expand from there, perhaps by having a virtual address that is translated with a <DCC++> command or a script in myAutomation.h. You could, for example, use the push button on the encoder to cycle through 3 locos and use the display on the CS, or a serial monitor to show you the change.

flash62au, 2022-04-25:
is this of any interest?
https://youtu.be/5gH5Dc0vIiU
https://github.com/flash62au/WiTcontroller

BT wiThrottle controller. 
Only three components.  the ESP32, keypad and Encoder.

closed 2025-03-08 by peteGSX