DCC-EX/CommandStation-EX #249: Feature request : integrate S88 feedback to DCC-EX

bereldhuin, 2022-06-28:
Hi,

Is there any plan to integrate S88 bus in DDC-EX as this is widely used by railroad modelers ? 

Some implementation have already been done by french people : 
http://lormedy.free.fr/DCCpp.html (this is a french page but I can help translate)

The S88 library is here : http://lormedy.free.fr/DCCppS88.zip

Fred

hansSchall, 2022-08-15:
I personally use SN74HC165 shift registers, they are used in many s88 devices.

I am currently devloping an arduino-based software connecting them to I2C. I will publish it, when it is production-ready.

Additionally I use SN74HC595 shift registers, which work exactly as SN74HC165 exept in the other direction so they are used as outputs.

Both togerther work as **very** cheap IO expander (0,50€ per 8 IO), but you have to add a dedicated microcontroller for frequent polling. I think integrating this into DCC-EX is not possible or **very** slow.

Asbelos, 2022-12-26:
We currently have a shift-register implementation of a HAL driver in our "devel" branch which is written for duinoNodes but may well be a model for similar shift-register based systems such as S88. 
If you have S88 hardware and would like to experiment with this code then see IO_duinoNodes.h in out devel branch.

Neil-McK, 2023-02-16:
I've experimented with creating an S88 driver for the HAL framework in DCC-EX.  However, the main issue I found was that the S88 bus can be very slow.  I believe that the original S88 devices were based on hardware shift registers (like the 4014 shift register).  These are rated for clocks up to 1MHz or more.  The duinoNodes implementation referred to above uses a nominal 2us clock period to read and write the S88 bus, allowing 128 inputs to be read/written in about 256us.

However, according to http://www.s88-n.eu/s88-timing-en.html, some modern S88 bus devices require over 200us clock period to work reliably, and many require 100us.  Since the DCC-EX CS is controlling locos on the track as well as performing other I/O to devices and sensors, the time that is allowed for a single 'loop'() call to complete in a device driver limited.  One might expect to be able to shift all of the inputs of an S88 bus within one 'loop()' entry, but each bit requires at least 200us on the above figures, and so a modest 16-bit S88 bus would require 32ms to poll, during which time the CS is unable to do any other functions.

An alternative approach would be to use a timer interrupt to operate the S88 clock and read/write the clocked data; this would be more effective but makes it more complex to implement, partly because of platform differences.  As I didn't have S88 hardware anyway I didn't proceed any further at the time.  We might need a straw poll to see if there's enough demand to make it worth while and what the scope would be - e.g. do we support the slow devices or only fast ones?

bask185, 2024-01-05:
I know this topic is 11 months old. But if the `micros()` function is still operational, you should use that one. Than you won't block your main loop() and `micros()` works on all platforms. I do really think you should add S88 because it is cheap and widely available as either DIY project or commerical products.

Flo100, 2024-11-16:
[https://github.com/DCC-EX/CommandStation-EX/pull/166#issuecomment-1319432784](url)
It can be implemented with myHal.cpp