DCC-EX/CommandStation-EX #278: [To Do]: Validate if native DCC-EX I2C implementation on Nucleo resolves device not detected issue

peteGSX, 2022-12-14:
### Task description

With OLED disabled in config.h and the current implementation of I2C for Nucleo which is using the Wire() library, EX-IODetector is unable to be detected on the I2C bus, despite a device at the same address being detected.

When OLED is enabled (even if not connected), this issue goes away.

Need to validate if this is a quirk of the Wire() library by testing again once the native I2C driver for Nucleo is implemented.

pmantoine, 2022-12-24:
Adding questions I've thought of so we don't lose them from my stoopid brain:

Does the Nucleo hang for a long time probing the I2C when this happens? Are there pullups on the EX-IO board for I2C? What values are they? Does this happen regardless of EX-IO board used... Nano/UNO/Mega etc.?

peteGSX, 2023-01-17:
This is a non-event. This was actually a bug in the device driver not calling I2CManager.begin(), in addition to a clock speed issue.

closed 2023-01-17 by peteGSX