DCC-EX/CommandStation-EX #279: Kato FL12

jerryshenming, 2022-12-23:
### Version

DCC-EX

### Issue description

I can program address for Kato motor DCC decoder, but I can not setup the address for Kato FL12 DCC light decoder,  is it compatible problem, what can I do? thanks!

### Hardware

Mega 2560 wifi UNO R3 mega328p ESP8266
L298 motor shield

habazut, 2022-12-23:
It could be that the FL12 still is "stuck in last century" and requires paged mode programming which DCC-EX CS can not do. Some things I read on the Internet sugest this. This would be the first decoder I know of sold in 2022 which can not do direct byte programming but considering what I have heard about the FL12 I would not be surprised. The  ZIMO MX605SL is a drop in replacement for the Kato (OEM Digitrax?) FL12. Zimo decoders can of course direct byte programming.

Regards,
Harald.

alex-code, 2022-12-23:
Is it just the write success that's failing?

If you write an address and move it back to the main track does the new address work?
Wondering if the CS says the address write fails as the decoder can't be read to verify?

jerryshenming, 2023-04-11:
> 

FL11 and FL12 does not send reply, it need page mode programming.

VR-Architect, 2023-12-18:
I just bought two new Kato locos and installed ESU Lok DCC 5v sound decoders. I had the exact same issue and searched for hours for a solution. My lights worked, but no movement. Then out of chance, I enter loco number 3 without any zeros in front and magically everything worked. So "03" and "003" are treated different than "3" in NCE Power Cab. Not sure this will fix your issue, but give it a try.

peteGSX, 2025-03-08:
@jerryshenming did the suggestion from @VR-Architect solve your issue?

jerryshenming, 2025-03-08:
> [@jerryshenming](https://github.com/jerryshenming) did the suggestion from [@VR-Architect](https://github.com/VR-Architect) solve your issue?

I built a program that finally solved this problem, instead of bought an old model Kato programmer.
https://github.com/jerryshenming/write-kato-FL11-12-address-id

peteGSX, 2025-03-08:
Ah I see, well done!

closed 2025-03-08 by peteGSX