DCC-EX/CommandStation-EX #300: [Feature Request]: throttle with output function

stephane-roux, 2023-02-01:
### Problem/idea statement

First, I would like to say bravo for this tools ! 
When I worked on it, I was not stopping saying whaou !

Among all the job you've done, no hurry, but I would like to be able to use a button to manage output.
I would need a button to run <Z 1 1> and <Z 1 0> (I assume I already prog <Z 1 53 00> for instance)

Idem for wifi throttle, but I've to lean Android Studio.

Have a nice time
Thanks again for your huge work
Stef

### Alternatives or workarounds

DCCpp CAB (Android), nearly do the job

### Additional context

n/a
If I can help you, translating, testing,...
My config is 
- Arduino Mega with Wifi
- Arduino motorshield
- AT firmware installed like a charm (super tuto)
- IDE installation (compiled without errors, uploaded configuring Arduino Swtiches)
- connecting to WiFi thru internet Box
Working like a charm, I needed 30min to install
\o/

habazut, 2023-02-01:
If you use the lastest version from devel you can define a route with exrail that will show up as a button in enginedriver. Look for the exrail documentation on our web site.  In exrail you manipulate pin numbers directly (53 in your example).

Harald.

habazut, 2023-03-27:
Look for the lastest changes in ED doing native DCCEX commands and contact the ED folks for feedback on that.
Harald.

closed 2023-03-27 by habazut

flash62au, 2023-03-27:
I recommend that you discuss this on the wifi-throttle channel on Discord.
https://discord.gg/PcvucXyY