DCC-EX/CommandStation-EX #307: Bug Report: EX-RAIL omits settings CV4 when using speed(0) in AUTOMATION(id)

AlbertG53, 2023-02-16:
### Version

4.1.2

### Bug description

SPEED(0) stops the locomotive without the delay in CV4 as in an emergency break.
And it only appears in AUTOMATION(id), not in SEQUENCE(id).

### Steps to reproduce the bug

When I use this:
SEQUENCE(12) 
  FWD(30)
  AT(22) SPEED(0)
The locomotive brakes according to the setting of CV 4, so with a little delay to 0 as it should be.

When I use this with Engine Driver to start the same locomotive:
AUTOMATION(12, "From 1 to 2")
  FWD(30)
  AT(22) SPEED(0) 

The locomotive brakes as with an emergency stop, so no delay according to CV4. The function (F4) on the locomotive to deactivate CV4/ABV is switched off.
Can anyone give a solution to this problem?
(When I drive the locomotive with the throttle and press Stop it stops with the delay according to CV4) 

### Expected behaviour

The locomotive should brake smoothly according to the setting in CV4

### Hardware in use

Arduino Mega + standard motor shield, no specific brand.

Asbelos, 2023-02-16:
The speed setting 0 in exrail sends the 0 stop message and has no knowlege of cv4, functions or whether the script was started by Ed... 

What is after SPEED(0) in your script?

Asbelos, 2023-02-16:
If your automation reaches a DONE statement or the task is terminated in any other way, that will estop your loco..  
You can avoid this by changing the current loco on the task, (SETLOCO(0) ) or using a delay

AlbertG53, 2023-02-17:
I posted this problem on Discord. The respons was it is a bug in EX-RAIL and I had to create an Bug Report. So I did.
I wonder why noboby on Discord came up with this answer.
This was my script with the problem:
AUTOMATION(12, "From 1 to 2")
FWD(30)
AT(22) SPEED(0)
DONE

I changed the script according to your message in
AUTOMATION(12, "From 1 to 2")
FWD(30)
AT(22) SPEED(0)
DELAY(1000)
DONE

And the problem is solved.
I think your answer of yesterday should be documented on the EX-RAIL pages

> If your AUTOMATION reaches a DONE statement or the task is terminated in any other way, that will estop your loco..
> You can avoid this by changing the current loco on the task, (SETLOCO(0) ) or using a delay
>

Asbelos, 2023-02-18:
My assessment of this problem was based on an assumption about what you didn't tell us on discord or in  this bug report... 
For future reference, its better to give all the info rather than jump to conclusions.

closed 2023-02-18 by Asbelos