DCC-EX/CommandStation-EX #314: Bug report:DCC-EX wifi can not start automatically

TomShen1234, 2023-02-28:
### Version

4.14

### Bug description

WIFI does not automatically connect to wifi router when system turn on, but if I connect PC to Arduino serial port, open Arduino IDE, connect correct port, wifi will auto start.

I did not nothing with in Arduino IDE

### Steps to reproduce the bug

1.turn on command station
2. can not ping Dcc-ex
3. connect PC to command station with serial port, open Arduino IDE, setup correct port
4. wifi then starting

### Expected behaviour

wifi can automatic start when power on

### Hardware in use

Arduino MEGA 2560 wifi

habazut, 2023-02-28:
I guess that when you connect the serial the Arduino Mega does a reset. Can you wake the wifi by just pressing the reset button. Either later or by holding the button 5 seconds while applying power?

Harald.

TomShen1234, 2023-02-28:
I found Mega2560 is not working fine with my ASUS ax86u router, it can stably work fine with Asus ac-5300

TomShen1234, 2023-02-28:
> I guess that when you connect the serial the Arduino Mega does a reset. Can you wake the wifi by just pressing the reset button. Either later or by holding the button 5 seconds while applying power?
> 
> Harald.

Thanks for your remind that every time when I plug in serial table, Mega is reset, after that I found  my mega is not working stable with Asus ax86u 2.4G band.

FrightRisk, 2023-03-01:
Is this what is usually called the Mega+WiFi? Both on one board? We have seen some reliability issues with that board. Many work, but some do not. We have not had issues with the ESP01s or the WiFi shields we recommend. But thank you for the report and for letting us know that some of the issues could be communication with specific routers

TomShen1234, 2023-03-13:
Yes, that is correct, it's the combined board.

closed 2023-03-13 by TomShen1234