DCC-EX/CommandStation-EX #322: [Feature Request]: implement on arduino giga

Maury49, 2023-03-16:
### Problem/idea statement

i would like to know if DCC++ runs on new ARDUINO Giga. it run on arduino uno and mega: can i hope is also running on arduino giga? ty very much

### Alternatives or workarounds

arduino uno and motor shield

habazut, 2023-03-16:
Not soon.

* In spite of the name and the look of the connectors, the HW has almost no features in common.
* No Arduino core in place yet
* Very expensive compared to what features DCC-EX really can use from that hardware.
* Someone who has "the itch" to port the software has to "scratch" it actually to do it.
* Other platforms like STM32 and ESP32 have much more promising price/performance and porting was already on it's way before that giga was even known.

Of course there is nothing that **prevents** a port. So if you or someone else is willing and can do it ....

Harald.

pmantoine, 2023-03-16:
Arduino have abandoned their old core library support on all STM32 based processors. They have instead switched to Mbed OS, and porting DCC-EX to that would be a great deal of work. So, we are using STM32duino core libraries for the port of DCC-EX to STM32F4xx processors, as that is a good deal less work (though still quite a lot!) The STM32F4xx Nucleo range will be our next generation of platform.

The Arduino Giga R1 uses an STM32H747XI processor, which is in the same overall ARM Cortex based family. However, at present STM32duino core only notionally supports this processor in a form that does not seem highly utilised. There is a "generic" style variant file support at present for the class of processor but NO specific board support. So this would represent yet more effort for our volunteer dev team... and more importantly right now, more difficult access and support for our users. In short the platform isn't ready for prime time.

As DCC-EX is already being ported and very actively tested on STM32F4xx hardware that is both more capable IO wise than the Arduino Giga, AND 1/8th-1/3rd the price of a Giga with direct board support from STM32duino's core libraries... you would be better served buying a Nucleo-F411RE, NucleoF446RE, or indeed a Nucleo-F429ZI or F429ZI at present. All these platforms accept Arduino UNO R3 style shields, and have up to 107 available GPIO pins which is more than the Giga. Prices range from $US13.50 to $US29 for very high quality boards from ST Microelectronics and are available globally.

Please see this link for updates on what we're already doing to go far beyond Mega support:

https://dcc-ex.com/under-development/microcontrollers.html

pmantoine, 2023-03-16:
We will, however, endeavour to support the Giga when support for it matures... but frankly it has features that so far give no advantageous features to even the most diehard DCC-EX user and is too expense for most of our users who are budget conscious as well.

peteGSX, 2025-03-08:
Just FYI the decision was made in July 2023 there will be no support for either Uno R4 or the Giga: https://dcc-ex.com/news/posts/20230728.html#gsc.tab=0

closed 2025-03-08 by peteGSX