DCC-EX/CommandStation-EX #329: Throttle will not reconnect after turning off then back on.

Budman1758, 2023-04-10:
### Version

Latest release

### Bug description

I don't know if this is a DCC-EX "thing" or a problem with the throttle itself. Using [EX-T3-WiFi ](https://github.com/DCC-EX/EX-T3-WiFi/issues/87)throttles. When I turn on the DCC-EX power supply and then turn one of my throttles on everything works as it should. I can turn the main and program tracks on and off and all other functions work fine. I then turn off the throttle and set it aside for a while leaving DCC-EX on. When I come back and turn the throttle back on it does not connect to the power supply. The icon showing connection to the PS shows connected but no functions work. After a short time the connection changes to showing disconnected.

BOTH of the throttles I currently have working show this behavior. Only way to get throttles to reconnect to the power supply is to power cycle it.

### Steps to reproduce the bug

1. Turn DCC-EX on.
2. Turn on the throttle and it connects and functions. Turn either main or program tracks on. (or both)
3. Turn throttle off leaving DCC-EX in track on state.
4. Come back after a bit, turn throttle back on.
5. Throttle does not connect to or control DCC-EX.

### Expected behaviour

Throttle should re-connect and function.

### Hardware in use

DCC-EX build as per basic recommended hardware
Throttles built the same.
This report also posted on the [throttle Github ](https://github.com/DCC-EX/EX-T3-WiFi/issues/87)page.

Budman1758, 2023-04-10:
Please see this issue in the [EX-T3-WiFi](https://github.com/DCC-EX/EX-T3-WiFi/issues/87) issue tracker. Pretty sure this is a Commandstation issue now.

Ash-4, 2023-04-11:
Has the DCC-EX WiFi ESP8266 been flashed to version 1.7.4 ?
Is the Mega powered using 7-9 volts to the barrel connector?
Attach the startup logs and diags from the Arduino serial monitor.
For additional diags, use the commands
<D WIFI 1> and/or <D WIT 1>

gflillich, 2023-04-11:
I am having a similar issue on a brand new build.  I am using wiThrottle 3.6 on a iphone as the throttle.  Commandstation hardware is:  elegoo arduino mega, arduino motorshield, and Makerfabs ESP8266 WIFI ShieldAfter initially powering up the Ardiuno I am able to connect, but, if I close out WiThrottle and try to reconnect it does not find the Commandstation.  I have to power cycle the commandstation.  I have tried the following:

1.  Powered via USB and 9V barrel.  Made no difference.
2.  Replaced Makerfab wifi shield with another.
3.  Tried in AP and station mode
4.  Reflashed several times
5.  Monitored the arduino via serial and can see commands, but, once I disconnect withrottle there are no commands
6.  Ping to the commandstation IP does not drop at any time.

I have not flashed the makerfab wifi shield.  I will attempt that next.

Ash-4, 2023-04-11:
Found a recent guide.
https://gist.github.com/nathankellenicki/7008540322c617869cec17226cff579d

There are other wiring options if you do not have the USB/TTL adapter.

habazut, 2023-04-12:
We need to move this into a prominent space on the DCC-EX web page. Do you know what the non-working firmware version was? Can you find that in an old log? My guess would be 2.something but what something.

Harald.

Budman1758, 2023-04-12:
> We need to move this into a prominent space on the DCC-EX web page. Do you know what the non-working firmware version was? Can you find that in an old log? My guess would be 2.something but what something.

Sorry but I do not have any old logs or anything else to show the old version.

gflillich, 2023-04-12:
My USB to TTL adapters are showing up today.  I will check the old firmware before I update it.  I have two shields to do.

Budman1758, 2023-04-19:
Unless someone needs it to remain open methinks this can be closed.

gflillich, 2023-04-19:
Flashing the firmware fixed the issue on both of my brand new wifi shields.  I was not able to find the current version of the firmware they had.