DCC-EX/CommandStation-EX #342: [Support Request]: Unable to use Analog pin A6 and A7 as sensors on Arduino Nano

bereldhuin, 2023-07-31:
### Version

4.1.6 (or even with 4.2.65)

### Issue description

Hi,

I can use all analog pins from my Arduino Nano (clone) as sensors (using pin A0 to A5 (#14 to 19)) but I'm not able to use A6 and A7 (which should be 20 and 21).

I know that on Uno, there are only 6 analog pins whereas Nano has 8.

These pins are working fine on a simple analogRead() Arduino sketch.

Is there any way to use them ? 

### Hardware

Arduino Nano clone, not using any Motor shield (but selected STANDARD_MOTOR_SHIELD in Installer).

peteGSX, 2023-07-31:
Pins A6/A7 have no digital capability at all and are analogue only, so if you're trying to use these as digital sensor pins it's actually not possible on the hardware at all.

bereldhuin, 2023-07-31:
Thank you for your reply.

I was not aware of this limitation. I tried  with a simple digitalRead() and as you stated it doesn't work either. 

I'm using a simple Arduino Nano with DCC-EX to only get turnout feedback sensors to Rocrail (DCC commands are done through an Intellibox).
I need 18 pins, and Nano "only" has 17 working with DCC-EX. So that's why I was trying to get A6 and A7 working.  ;-)

peteGSX, 2023-07-31:
Ah yes, initially I also thought great, Nano has two extra pins! Unfortunately, not quite in the digital sense. They do actually display them as analogue only on the datasheet, but it's easy to miss, especially as all other analogue pins work as digital pins also.

closed 2023-07-31 by peteGSX