DCC-EX/CommandStation-EX #357: Incorporate Track Modes into OLED Display

twr140, 2023-10-01:
At present there isn't a built in means to have an OLED connected to the system display the Track Mode selected for each track output.

While it is possible to create Automations / Routes in EX_Rail, and have these display track states using the SCREEN() function, if the track mode is changed via serial port commands or Engine Driver's Track Manager, the OLED display is not updated for the track mode, and is then not in sync with the actual track mode and hence shows the wrong information.

It would be beneficial to have the ability to show these track modes on systems that have their track modes changed regularly, such as systems used for Public Displays where DC or DCC trains might be run on any given track output at any time.

ColinHMurdoch, 2023-10-11:
Added returnMode(byte t),returnDCAddr(byte t) and getModeName(byte Mode) to TrackManager which allowed the use of a UserAddin function to display the required information to an OLED/LCD screen.

ColinHMurdoch, 2023-10-11:
User tested the UserAddin function with success.

closed 2023-10-11 by ColinHMurdoch