DCC-EX/CommandStation-EX #367: [Feature Request]: Add EEPROM Functionality to ARDUINO_ARCH_ESP32

schwefeljm, 2023-11-12:
I have worked on this and have a branch locally I would like to push upstream to get feedback on. 
It turned out to be pretty simple and allows for up to 64k of EERPROM storage.

Please advise on how I should proceed.

Thank you,

Jason

habazut, 2023-11-12:
Fork the CS repo to your github and then push your local branch there and we'll have a look at it.

It is probably simple. If I remember correct, on ESP32 one needs to commit all changes to EEPROM in a seperate call as writes do not immidiately get copied to persistent storage.

Harald.

schwefeljm, 2023-11-12:
@habazut - Git is saying "You do not have permissions to push to 'DCC-EX/CommandStation-RX' on Github"

habazut, 2023-11-12:
Of course not. Fork out repo to **your** repo, add a new upstream definition from your local and push to your repo
```
git remote add myforkrepo git@github.com:schwefeljm/CommandStation-EX
git push myforkrepo 
```
Harald.

schwefeljm, 2023-11-12:
Understood. I am used to the model we use at work where (mostly) anyone can push a branch, but main is protected and requires PR, approvals, etc.

Here it is:
https://github.com/schwefeljm/CommandStation-EX/tree/esp32-eeprom