DCC-EX/CommandStation-EX #375: Ethernet not obtaining IP address

VR-Architect, 2023-12-25:
I have swapped out my working WIFI board for the Arduino Ethernet 2 board. I used EX-Installer and set the config file with my WIFI-SSID and WIFI-PASSWORD per the instructions. I have a good network connection per indicator lights. The display shows IP as 0.0.0.0 and I don't see the device on my network.

Questions:
- Why do we need to populate the SSID if it is a wired connection?
- Does it use DHCP automatically?
- When using the WIFI card, I need a wire for transmit and another for receive. Do we need the same for the Arduino Ethernet 2 board? If yes, I was not able to find instructions.
- I noticed the instructions state a max of 4 WIFI connections and there seems to be limitations on the Ethernet 2 connections also. So how would a club with say 15 people connected at the same time use DCC-EX?

Thanks for the help.

Ash-4, 2023-12-25:
The WiFi settings would not be needed with Ethernet connection.

Check for these lines in config.h  
#define ENABLE_WIFI false
#define ENABLE_ETHERNET true

The Ethernet shield must be the first shield on the Mega, because the Ethernet shield uses the pins in the ICSP header.
Check to see that pins 4, 10, 50-53 are not defined or connected for other use.

EX-Installer should provide the library needed, but if the Arduino IDE install method is used:
...  you want the Ethernet library 2.0.0 or 2.0.1 from the Arduino IDE and NOT the one named Ethernet2 which spooks around the Internet.

Ethernet + router should provide for more connections than WiFi.  (someone else can advise on this.)

habazut, 2023-12-25:
>     * Why do we need to populate the SSID if it is a wired connection?

You don't. Choose Wifi or Ethernet but not both in config.h

>     * Does it use DHCP automatically?

Yes.

>     * When using the WIFI card, I need a wire for transmit and another for receive. Do we need the same for the Arduino Ethernet 2 board? If yes, I was not able to find instructions.

The board communicates with a whole bunch of wires and thus must sit direct on the Arduino.

>     * I noticed the instructions state a max of 4 WIFI connections and there seems to be limitations on the Ethernet 2 connections also. So how would a club with say 15 people connected at the same time use DCC-EX?

Yes, there is a limit as well, probably dependent on the Ethernet library and/or amount of RAM you have.

If you want to connect as much as 15 withrottles I recommend to use the JMRI withrottle server running on a computer with much more memory than an arduino and then connect the arduino via USB.

Harald.

VR-Architect, 2023-12-25:
Who can update this link for the correct settings? This is the one I followed, and it seems to be incorrect - thus all the weird questions. 

https://dcc-ex.com/reference/hardware/ethernet-boards.html#gsc.tab=0 

I followed @Ash-4 instructions and it works great. It might also help neebies if someone could update this link also:

https://dcc-ex.com/reference/hardware/ethernet-boards/arduino-ethernet-shield-2.html#gsc.tab=0 

@habazut Thanks for your response. I got JMRI installed and working. Maybe in the future the requirement for JMRI could be removed with better Ethernet cards.

Thank you both for the fast help!